zephyrproject-rtos/zephyr #4165: ieee802154_uart_pipe.c: warning: return from incompatible pointer type

pfalcon, 2017-10-03:
Building echo_client with make "CONF_FILE=prj_qemu_802154.conf", there's a warning:

~~~
  CC      drivers/ieee802154/ieee802154_uart_pipe.o
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr/drivers/ieee802154/ieee802154_uart_pipe.c: In function ‘upipe_rx’:
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr/drivers/ieee802154/ieee802154_uart_pipe.c:96:9: warning: return from incompatible pointer type [-Wincompatible-pointer-types]
  return pkt;
         ^~~
~~~

I quickly compared that with e.g. slip.c, and saw processing in rx callback is quite different there. @tbursztyka , can you please see if it requires just a cast, or more things bitrotted there?

pfalcon, 2017-10-03:
To see this warning (instead of bunch of errors), https://github.com/zephyrproject-rtos/zephyr/pull/4166 should be applied first.

pfalcon, 2017-10-03:
@priyankaNXP: FYI

tbursztyka, 2017-10-04:
Good catch, it's a net_nbuf/net_pkt introduced error (commit id db11fcd174). So instead of pkt it should return buf as before. Will fix it asap

closed 2017-10-04 by pfalcon