zephyrproject-rtos/zephyr #4182: NET_APP_SETTINGS for 15.4 doesn't seem to work (if to trust 15.4 shell)

pfalcon, 2017-10-04:
I'm starting to play with 15.4 on BOARD=frdm_kw41z (didn't have previous 15.4 hands-on experience). I build echo_server with https://github.com/zephyrproject-rtos/zephyr/pull/4181 which takes care of enabling a driver and related settings, including 15.4 NET_APP_SETTINGS. Additionally, I enable CONFIG_NET_L2_IEEE802154_SHELL. In outdir/frdm_kw41z/.config I have:

~~~
CONFIG_NET_APP_IEEE802154_DEV_NAME=""
CONFIG_NET_APP_IEEE802154_PAN_ID=0xabcd
CONFIG_NET_APP_IEEE802154_CHANNEL=26
CONFIG_NET_APP_IEEE802154_RADIO_TX_POWER=0
~~~

(which are the default settings).

But after startup and trying to check these params via 15.4 shell, they read back as zeros. However, I can change them via shell. To me, that looks as if NET_APP_SETTINGS don't really configure 15.4 params (vs for example a driver issue). @tbursztyka : Does anything like that sound familiar? Can you confirm that it works as expected on your hardware?

Note that I didn't try to check whether it "works" yet - I'm wading towards that thru darkness, so prefer to check each step.

pfalcon, 2017-10-04:
@dleach02, @priyankaNXP : FYI

tbursztyka, 2017-10-05:
Will check, I haven't been using NET_APP since its addition as I am 100% on lower level, the ieee802154 samples use  ieee802154_settings.c and that have always worked. That said net_app does basically the same so "it should" work. Actually the patch your are referencing is pointing out issues btw (sockets samples)

pfalcon, 2017-10-05:
Ok, I see the issue:

~~~
        dev = device_get_binding(CONFIG_NET_APP_IEEE802154_DEV_NAME);
        if (!dev) {
                return -ENODEV;
        }
~~~

So, CONFIG_NET_APP_IEEE802154_DEV_NAME should be set correctly for it to work, I'll update #4181. 

Keeping open until confirmed to work.

pfalcon, 2017-10-10:
So, this was a result of a combination of lack of proper default settings for a particular boards, and lack of error reporting by default (our 2 common problems).