zephyrproject-rtos/zephyr #4188: tests/net/tcp/test_tcp.py#_ipv4_tcp :Evalution failed

rrahulkx, 2017-10-05:
This testcase completion failed qemu-zephyr-x86 .

the Error log is attached to this .

[Compeltation_failure.txt](https://github.com/zephyrproject-rtos/zephyr/files/1358520/Compeltation_failure.txt)

nashif, 2017-10-05:
Please include the relevant parts of the log you have attached to the issue directly, like so for example:

```
console output: ***** delaying boot 1000ms (per build configuration) *****
console output: ***** BOOTING ZEPHYR OS v1.9.99 (delayed boot 1000ms) *****
console output: shell> [net/app] [INF] net_app_init: Initializing network
console output: [net/app] [INF] setup_ipv4: IPv4 address: 192.168.106.30
console output: [net/app] [INF] ipv6_event_handler: IPv6 address: fe80::6a:ff:fe00:1e
console output: [net/app] [INF] new_client: Connection from [2001:db8::2]:42464 (0x00400428)
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 110 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 724 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 682 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 1208 bytes
console output: [net/app] [DBG] _net_app_accept_cb: (0x004001a0): Already connected via context 0x00400428
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 514 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 938 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 748 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 740 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 514 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 882 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 278 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 318 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 188 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 898 bytes
console output: [echo-server] [INF] build_reply_pkt: TCP IPv6 received 1104 bytes
console output: [net/app] [DBG] net_app_send_pkt: (0x004001a0): Cannot send to peer (-22)
console output: [echo-server] [ERR] tcp_received: ^[[0;31mCannot send data to peer (-22)^[[0m
console output: [echo-server] [INF] main: Stopping...
console output: ***** CPU Page Fault (error code 0x00000000)
console output: Supervisor thread read address 0x00000004
console output: PDE: 0x025 Present Read-only User
console output: PTE: 0x000 Non-present Read-only Supervisor
console output: Current thread ID = 0x00401790
console output: Faulting segment:address = 0x0008:0x0000512c
console output: eax: 0xffffffff, ebx: 0x00000000, ecx: 0x000051b1, edx: 0x00400890
console output: esi: 0x00000000, edi: 0xaaaaaaaa, ebp: 0x004153c0, esp: 0x004153bc
console output: eflags: 0x286
console output: Fatal fault in thread 0x00401790! Aborting.
```

nashif, 2017-10-05:
You need to provide more details about how you are testing this, the title is also misleading, zephyr does not have `tests/net/tcp/test_tcp.py`, so someone not familiar with the test env you are using will not know what you are talking about it.

gurucharanx, 2017-10-06:
The issue needs additional configuration settings of CONFIG_NET_APP_NEED_IPV4=y and CONFIG_NET_APP_NEED_IPV6=n. This resolve this issue

nashif, 2017-10-06:
> The issue needs additional configuration settings of CONFIG_NET_APP_NEED_IPV4=y and CONFIG_NET_APP_NEED_IPV6=n. This resolve this issue

Ok, then please close it if it is resolved...

gurucharanx, 2017-11-08:
Configuring the CONFIG_NET_APP_NEED_IPV4=y or CONFIG_NET_APP_NEED_IPV6=y along with CONFIG_NET_IPV4=y or CONFIG_NET_IPV6=y solves only the problem of connection between the sockets in the particular domain. But still facing the issue of packets transfer between the sockets.

rrahulkx, 2017-11-24:
This is Duplication of following github
 https://github.com/zephyrproject-rtos/zephyr/issues/4216
closing this Issue .

closed 2017-11-24 by rrahulkx