zephyrproject-rtos/zephyr #4193: Zephyr libc(snprintf) is not comply with ISO standard.

vikramx2112, 2017-10-05:
Tried different combination of integer and floating point format specifier with width specifier on snprintf,  below is the result with GNU libc (running on linux host machine(ldd (GNU libc) 2.24)) and zephyr minimal libc.
Precondition:
- snprintf  buf is an array of 2000 bytes.
char buf[2000];

1) INTEGER based format specifier:
(a)WITH GNU libc
i. snprintf(buf,sizeof(buf),"%.0d",0)
return 0, and buff contains "" (null string).

ii.snprintf(buf,sizeof(buf),"%.0o",0)
return 0, and buff contains "" (null string).

iii. snprintf(buf,sizeof(buf),"%#.0d",0)
return 0, and buff contains "" (null string).

iv. snprintf(buf,sizeof(buf),"%#.0x",0)
return 0, and buff contains "" (null string).

v. snprintf(buf,sizeof(buf),"%2.0d",0)
return 2, and buff contains "  ".

(b) WITH ZEPHYR minimal libc
i. snprintf(buf,sizeof(buf),"%.0d",0)
return 1, and buff contains "0".

ii.snprintf(buf,sizeof(buf),"%.0o",0)
return 1, and buff contains "0".

iii. snprintf(buf,sizeof(buf),"%#.0d",0)
return 1, and buff contains "0".

iv. snprintf(buf,sizeof(buf),"%#.0x",0)
return 3, and buff contains "0x0".

v. snprintf(buf,sizeof(buf),"%2.0d",0)
return 2, and buff contains " 0".

2) Floating point based format specifiers:

(a) With GNU libc

i. Format specifier %a is working with GNU libc as expected. but with Zephyr libc %a is always returning 0, and buff contains null string.

ii. Conversion of large integer.
snprintf(buf, sizeof(buf),"%.0f",340282366920938463463374607431768211456.0)
return 39 and buf contains "340282366920938463463374607431768211456"

iii. snprintf(buf, sizeof(buf),"%e",0.0)
return 12, and buf contains "0.000000e+00"

iv. snprintf(buf, sizeof(buf),"%.4f",1.03125)
return 6, buf contains "1.0312"

v. snprintf(buf,sizeof(buf),"%g",1234567)
return 11, buf contains "1.23457e+06"

(b) With Zephyr Libc

i. Format specifier %a is working with GNU libc as expected. but with Zephyr libc %a is always returning 0, and buff contains null string.

ii. snprintf(buf, sizeof(buf),"%.0f",340282366920938463463374607431768211456.0)
return 39 and buf contains "340282366920938500000000000000000000000"

iii. snprintf(buf, sizeof(buf),"%e",0.0)
return 13, and buf contains "0.000000e+000"

iv. snprintf(buf, sizeof(buf),"%.4f",1.03125)
return 6, buf contains "1.0313"

v.snprintf(buf,sizeof(buf),"%g",1234567.0)
return 7, buf contains "1234567"

pfalcon, 2017-10-05:
An obvious way forward is try newlib (CONFIG_NEWLIB_LIBC) instead of Zephyr's minimal libc, it's called minimal for a reason.

vikramx2112, 2017-10-06:
Thanks @pfalcon, by adding CONFIG_NEWLIB_LIBC, solved issues related to ISO compliance.

But for %a format specifier seems like there is no support available in Zephyr. Verified %a failure using below instruction.
snprintf(buf,sizeof(buf),"%a",0.0);
Running with Zephyr NEWLIB libc: buf contains "a" instead of "0x0p+0".

Is there any plan to add support of %a in zephyr libc.

pfalcon, 2017-10-24:
@vikramx2112:

The answer is two-sided:

1. As mentioned, Zephyr uses separate project, newlib, as libc, so if "%a" will be implemented in it, Zephyr might get support for it too eventually.
2. What's the usecase for that format specifier? Zephyr's purpose is to be OS for small IoT devices, starting from 8K of FlashROM. How having "%a" specifier helps to achieve that aim? Its effect would be the opposite - the more features there're, the more code it requires (which would lead to inability to use Zephyr on more resource-constrained devices).

vikramx2112, 2017-10-25:
@pfalcon , we are adding some testcode to test libc feature, there i seen that newlib is not supporting %a format specifier. That is a reason i asked to know whether is there any plan to add this one or not.
I understand that more code will increase the memory footprint that will affect system performance as well as it can make code to overflow to device memory.
Closing this as main intention of ISO comply issue is resolved.

closed 2017-10-25 by vikramx2112