zephyrproject-rtos/zephyr #4213: samples/net/: unexpected communication happens between echo_client and echo_server sample applications

gurucharanx, 2017-10-06:
When I tried to communicate between the echo_server sample application and echo_client sample application with the below settings it communicates wrongly.
Below are configuration settings for server and client respectively 
**At the server end :**
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_UDP=n
CONFIG_NET_TCP=y
CONFIG_NET_APP_SERVER=y
CONFIG_NET_APP_NEED_IPV6=y
CONFIG_NET_APP_NEED_IPV4=y
CONFIG_NET_APP_SETTINGS=y
CONFIG_NET_APP_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_APP_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_APP_MY_IPV4_ADDR="192.168.2.1"
CONFIG_NET_APP_PEER_IPV4_ADDR="192.168.2.2"

**At the client end :**
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_UDP=n
CONFIG_NET_TCP=y
CONFIG_NET_APP=y
CONFIG_NET_APP_NEED_IPV6=y
CONFIG_NET_APP_NEED_IPV4=y
CONFIG_NET_APP_CLIENT=y
CONFIG_NET_APP_SETTINGS=y
CONFIG_NET_APP_MY_IPV6_ADDR="2001:db8::2"
CONFIG_NET_APP_PEER_IPV6_ADDR="2001:db8::1"
CONFIG_NET_APP_MY_IPV4_ADDR="192.0.2.2"
CONFIG_NET_APP_PEER_IPV4_ADDR="192.0.2.1"

The above settings are made to verify whether the applications can communicate in IPv4 and IPv6 if  they have been set with some wrong IP addresses(which is a negative test case). As it is not expected to communicate in either IPv4(as given wrong IP addresses) nor in IPv6(configuration is unset)
As per the above settings I am in a impression that it shouldn't communicate in both IPv4 and IPv6 domains. But these two applications communicating in IPv6(which shouldn't expected to happen) for sometime suddenly giving the error as cannot connect TCP and cannot init IPv4 TCP which I was expecting at the begining before packet transfer 
Below are the respective logs 

**At the server:**

shell> [dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[dev/eth_mcux] [INF] eth_mcux_phy_event: Link down
[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 1042 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 84 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 1008 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 574 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 7 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 795 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 744 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 629 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 494 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 826 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 930 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 1021 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 1229 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 919 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 997 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 80 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 254 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 659 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 140 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 629 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 839 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 1120 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 483 bytes
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 917 bytes
.
.
.
.
.
.

**At the client** 

shell> [dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[echo-client] [INF] init_app: Run echo client
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 1042 bytes, received 1042 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 1042 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 84 bytes, received 84 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 84 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 1008 bytes, received 1008 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 1008 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 574 bytes, received 574 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 574 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 7 bytes, received 7 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 7 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 795 bytes, received 795 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 795 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 744 bytes, received 744 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 744 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 629 bytes, received 629 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 629 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 494 bytes, received 494 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 494 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 826 bytes, received 826 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 826 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200019e0): Sent 930 bytes, received 930 bytes
.
.
.
.
.
.[echo-client] [DBG] tcp_received: (0x200019e0): Sent 980 bytes, received 980 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 980 bytes, all ok
[echo-client] [ERR] connect_tcp: Cannot connect TCP (-60)echo-client] [DBG] tcp_received: (0x200019e0): Sent 386 bytes, received 386 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 386 bytes, all ok
0m
[echo-client] [ERR] start_tcp: Cannot init IPv4 TCP clie[echo-client] [DBG] tcp_received: (0x200019e0): Sent 468 bytes, received 468 bytes
[echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 468 bytes, all ok
nt (-60)
[echo-client] [INF] main: Stopping...

~

jukkar, 2017-10-06:
> [echo-client] [DBG] compare_tcp_data: (0x200019e0): Compared 980 bytes, all ok
> [echo-client] [ERR] connect_tcp: Cannot connect TCP (-60)echo-client] [DBG] tcp_received: (0x200019e0): Sent 386 bytes, [...]

This looks like an impossible error.
So you first get lot of successful data transfers, and then the application tries to call connect_tcp() again. But this is just not possible as the echo-client calls connect_tcp() exactly once when it starts. So it looks like your log contains information about two echo-client invocation.

gurucharanx, 2017-10-09:
It seems,in the echo client application connect_tcp()  is invoked 2 times for IPv4 and IPv6 separately.  Just because of this in IPv6 the data packets were transferred even though they were not expected to be transferred because of  (CONFIG_NET_IPV6=n). Am I correct? 
 I have verified once again that the echo client is invoked only once.
The above error is coming every-time I run it.
Lastly I have verified on commit ID: 3e3a2379305317a06bb6ac48b3d4f8688155192d

jukkar, 2017-10-09:
> It seems,in the echo client application connect_tcp() is invoked 2 times for IPv4 and IPv6 separately

Yes, you are correct. Either IPv4 or IPv6 connectivity does not work in your example run. In your config, you should also set CONFIG_NET_APP_NEED_IPV6=n as if you set it to "y", then it will automatically activate IPv6.

jukkar, 2017-10-16:
Can you verify if everything works ok if you set CONFIG_NET_APP_NEED_IPV6=n in your tests (assuming of course that you want to test IPv4 only)?

jukkar, 2017-10-16:
I have been investigating this and found following:

If both IPv6 and IPv4 are enabled, then currently #4230 is needed in echo-server so that multiple incoming connections are support. After applying #4230 one also needs to set CONFIG_NET_TCP_BACKLOG_SIZE=2 and CONFIG_NET_APP_SERVER_NUM_CONN=2 options in server config file. If #4230 is not applied, then if client is rebooted after successfully transferring some data, then server needs to be rebooted too, otherwise server will not accept the new connection (because of one connection limit).

If frdm_k64f is used in the test, then if the ethernet MAC address is randomized (this is the default), then it might happen that if server is running, and one reboots the client, the server might still have wrong mac address in arp cache. This will prevent connectivity in this case. So one would need to reset the server too, or use static link address.
Static MAC address can be set by setting last three bytes like this:
```
CONFIG_ETH_MCUX_0_RANDOM_MAC=n
CONFIG_ETH_MCUX_0_MAC3=0x01
CONFIG_ETH_MCUX_0_MAC4=0x01
CONFIG_ETH_MCUX_0_MAC5=0x01
```

And as I said in earlier comments, if you want to only test IPv4, then you must set both CONFIG_NET_IPV6=n and CONFIG_NET_APP_NEED_IPV6=n in your config.

About this bug, I was able to reproduce the output in my tests. Usually everything worked ok, but depending on how the client and server were rebooted I could trigger the failure. The #4355 might help in here too, that PR will be applied soon.

gurucharanx, 2017-11-28:
@jukkar  I have tried to config  CONFIG_NET_IPV6=n and CONFIG_NET_APP_NEED_IPV6=n in my config for IPv4 but still getting the similar issues.

MaureenHelm, 2018-03-23:
@gurucharanx @jukkar Is this still an issue?

kumarvikash1, 2018-03-24:
@tarunkum , please update the issue status

tarunkum, 2018-03-27:
@jukkar , @kumarvikash1 

I tried to test with 1.11 RC3 tag. I did not find any issues related to communication between http_client on Zephyr target and http_server on local machine.

Please find the below logs from my machine.

window 1

Zephyr/samples/net/http_client

[http-client] [INF] response: HTTP body: 24 bytes, expected: 24 bytes
[http-client] [INF] do_async_http_req: [18] Send /big-file.html
[http-client] [INF] response: Received 553 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 90 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 1528 bytes, expected: 1528 bytes
[http-client] [INF] do_async_http_req: [19] Send /index.html
[http-client] [INF] response: Received 347 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 161 bytes, expected: 161 bytes
[http-client] [INF] do_async_http_req: [19] Send /
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] do_async_http_req: [19] Send /post_test.php
[http-client] [INF] response: Received 163 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 24 bytes, expected: 24 bytes
[http-client] [INF] do_async_http_req: [19] Send /big-file.html
[http-client] [INF] response: Received 553 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 90 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 1528 bytes, expected: 1528 bytes
[http-client] [INF] do_async_http_req: [20] Send /index.html
[http-client] [INF] response: Received 347 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 161 bytes, expected: 161 bytes
[http-client] [INF] do_async_http_req: [20] Send /
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] do_async_http_req: [20] Send /post_test.php
[http-client] [INF] response: Received 163 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 24 bytes, expected: 24 bytes
[http-client] [INF] do_async_http_req: [20] Send /big-file.html
[http-client] [INF] response: Received 553 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 90 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 1528 bytes, expected: 1528 bytes
[http-client] [INF] main: Done!

window 2

sudo net-tools/http-server.py 

::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /big-file.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /index.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "HEAD / HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "POST /post_test.php HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /big-file.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /index.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "HEAD / HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "POST /post_test.php HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /big-file.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /index.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "HEAD / HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "POST /post_test.php HTTP/1.1" 200 -
[... 8 lines cut ...]
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /big-file.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /index.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "HEAD / HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "POST /post_test.php HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /big-file.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /index.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "HEAD / HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "POST /post_test.php HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /big-file.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /index.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "HEAD / HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "POST /post_test.php HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /big-file.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:53] "GET /index.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "HEAD / HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "POST /post_test.php HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "GET /big-file.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "GET /index.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "HEAD / HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "POST /post_test.php HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "GET /big-file.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "GET /index.html HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "HEAD / HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "POST /post_test.php HTTP/1.1" 200 -
::ffff:192.0.2.1 - - [27/Mar/2018 12:19:54] "GET /big-file.html HTTP/1.1" 200 -

jukkar, 2018-05-02:
Closing this as everything seems to work ok.

closed 2018-05-02 by jukkar