zephyrproject-rtos/zephyr #4216: samples:net:sockets:echo : communications blocks between client and server after sometime

gurucharanx, 2017-10-06:
The communication between the server and client blocks after few packets transmission when i used the sample application of sockets/echo as a TCP server and samples/echo_client as a TCP client.
The configuration settings for Server and client are as below

**At Server:**
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_APP_SETTINGS=y
CONFIG_NET_APP_NEED_IPV4=y
CONFIG_NET_APP_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_APP_PEER_IPV4_ADDR="192.0.2.2"

**At Client:**
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_UDP=n
CONFIG_NET_TCP=y
CONFIG_NET_APP_SETTINGS=y
CONFIG_NET_APP_CLIENT=y
CONFIG_NET_APP_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_APP_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_APP_MY_IPV4_ADDR="192.0.2.2"
CONFIG_NET_APP_PEER_IPV4_ADDR="192.0.2.1"

The console o/p after the applications running at the server and client are as below

**At server:**
[dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:3e:2b:8a
[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[dev/eth_mcux] [INF] eth_mcux_phy_event: Link down
[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
Connection from 192.0.2.2

**At Client:**
[dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:02:d6:e6
shell> [dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[echo-client] [INF] init_app: Run echo client
[echo-client] [DBG] tcp_received: (0x200023e0): Sent 644 bytes, received 74 bytes
[echo-client] [DBG] compare_tcp_data: (0x200023e0): Compared 74 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200023e0): Sent 644 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200023e0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200023e0): Sent 644 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200023e0): Compared 128 bytes, all ok

pfalcon, 2017-10-06:
@gurucharanx : What are these tests which you run? Are they related to reports like https://github.com/zephyrproject-rtos/zephyr/issues/4195, https://github.com/zephyrproject-rtos/zephyr/issues/4190, etc. ? Why don't you use standard configs? Why do you try to test 2 things at once (echo server and echo client)?

jukkar, 2017-10-12:
Please provide exact instructions and config files you used so that other people can try to reproduce this issue.
For example the server config file you have here, the sockets/echo application does not even compile.

gurucharanx, 2017-10-12:
@jukkar , I have used the same configuration file as in the respective folders except adding/changing the above configurations in the respective files.

jukkar, 2017-10-12:
I managed to compile and run this with two frdm_k64f. And I could see the same issue, meaning that samples/net/socket/echo server and samples/net/echo_client do not seem to work properly when connected to each other.
Then I tried the same setup but using samples/net/echo_server and samples/net/echo_client. This seems to work and TCP data was transferred reliably between two devices.
Then I looked the socket/echo/src/socket_echo.c code and it is only receiving 128 bytes data and then sending it back. Because this is a TCP connection, it should not really matter what is the size of the receiving buffer in socket_echo.c but it looks like this is causing some issues here.
I investigated the wireshark output of the traffic and it seems the server received all the data (ACK for all the data is received in client), which in my test 1121 bytes long but only sent 330 back to client. The client just waits the remaining data which is never sent.
@pfalcon Have you seen anything like this before?

jukkar, 2017-10-12:
I figured out why the connectivity stops. The reason is that the server runs out of buffers and is not able to send any data. After I increased the amount of buffers, the connectivity went further but then stopped again. It seems that there is a buffer leak in sockets/echo application.

I also saw this message in server log:

[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 10M half-duplex mode.
Connection from 192.0.2.2
[net/buf] [ERR] net_buf_alloc_debug: net_pkt_get_reserve_data_debug():362: Failed to get free buffer
[net/pkt] [ERR] net_pkt_unref_debug: *** ERROR *** pkt 0x2000950c is freed already by prepare_segment():386 (zsock_sendto():274)

pfalcon, 2017-10-12:
@jukkar : I never tested echo_client + sockets/echo, because that goes against testing best practices, which say there should be one entity under test, and the rest of test environment should be well known working. So, I always test with one side being Linux. And I prefer well-known existing tools, not an adhocly developed ones, and my standard automated testcase is running "ab" (ApacheBench) on host against an HTTP server on Zephyr device. That used to work as expected, until https://github.com/zephyrproject-rtos/zephyr/issues/4022 hit. As mentioned there, my current investigation showed that there started to be a shortage of free packets in SLIP driver. With some fuzz factor applied, there can be similarity to the shortage of packets you saw.

Anyway, the error you quote is quite eloquent on its own, let me look into that first.

pfalcon, 2017-10-12:
> Anyway, the error you quote is quite eloquent on its own, let me look into that first.

Ok, so you get https://github.com/zephyrproject-rtos/zephyr/issues/4292 for that. And if you say above

> The reason is that the server runs out of buffers and is not able to send any data.

It may be the same issue.

Note that in #4022, my investigation showed that *receive* buffers for SLIP drivers are lacking. They also didn't look like leaked IIRC, but rather that after https://github.com/zephyrproject-rtos/zephyr/commit/84db641de66463550fb41f083dbdf0b02fc78284, these buffers were there with the lesser probability (or after higher turnaround time) than before that patch.

jukkar, 2017-10-13:
> Note that in #4022, my investigation showed that receive buffers for SLIP drivers are lacking. They also didn't look like leaked IIRC, but rather that after 84db641, these buffers were there with the lesser probability (or after higher turnaround time) than before that patch.

This issue has nothing to do with SLIP, as the test is using normal ethernet.

jukkar, 2017-10-13:
I am adding some debugging about the leak. I did not managed to follow the code flow yet, so still a mystery why the packets were not released after TCP ack is received.

```
Connection from 192.0.2.2
[net/buf] [ERR] net_buf_alloc_debug: net_pkt_get_reserve_data_debug():362: Failed to get free buffer

shell> net allocs
Network memory allocations

memory          Status  Pool    Function alloc -> freed
0x200092f0/1    used    TX      zsock_sendto():236
0x2000914c/1    used    TX      zsock_sendto():236
0x200094d0/1    used    TX      zsock_sendto():236
0x20009494/1    used    TX      zsock_sendto():236
0x20009458/1    used    TX      zsock_sendto():236
0x2000941c/1    used    TX      zsock_sendto():236
0x200093e0/1    used    TX      zsock_sendto():236
0x200093a4/1    used    TX      zsock_sendto():236
0x20009368/1    used    TX      zsock_sendto():236
0x2000932c/1    used    TX      zsock_sendto():236
0x2000e648/1    used    TDATA   net_pkt_append():1179
0x2000dd48/1    used    TDATA   net_pkt_append():1179
0x20008d50/1    used    TX      zsock_sendto():236
0x2000de68/1    used    TDATA   net_pkt_append():1179
0x2000def8/1    used    TDATA   net_ipv4_create_raw():37
0x2000df88/1    used    TDATA   net_pkt_append():1179
0x2000e018/1    used    TDATA   net_ipv4_create_raw():37
0x2000e0a8/1    used    TDATA   net_pkt_append():1179
0x2000e138/1    used    TDATA   net_ipv4_create_raw():37
0x2000e1c8/1    used    TDATA   net_pkt_append():1179
0x2000e258/1    used    TDATA   net_ipv4_create_raw():37
0x2000e2e8/1    used    TDATA   net_pkt_append():1179
0x2000e378/1    used    TDATA   net_ipv4_create_raw():37
0x2000e408/1    used    TDATA   net_pkt_append():1179
0x2000e498/1    used    TDATA   net_ipv4_create_raw():37
0x2000e528/1    used    TDATA   net_pkt_append():1179
0x2000e5b8/1    used    TDATA   net_ipv4_create_raw():37
0x2000e768/1    used    TDATA   net_ipv4_create_raw():37
0x2000e6d8/1    used    TDATA   net_ipv4_create_raw():37
0x200092b4/1    used    TX      zsock_sendto():236
0x20009278/1    used    TX      zsock_sendto():236
0x2000923c/1    used    TX      zsock_sendto():236
0x20009200/1    used    TX      zsock_sendto():236
0x200091c4/1    used    TX      zsock_sendto():236
0x20009188/1    used    TX      zsock_sendto():236
0x2000ee28/1    used    TDATA   net_pkt_append():1179
0x2000ddd8/1    used    TDATA   net_pkt_append():1179
0x20008f30/1    used    TX      zsock_sendto():236
0x2000e888/1    used    TDATA   net_pkt_append():1179
0x2000e918/1    used    TDATA   net_ipv4_create_raw():37
0x2000e9a8/1    used    TDATA   net_pkt_append():1179
0x2000ea38/1    used    TDATA   net_ipv4_create_raw():37
0x2000eac8/1    used    TDATA   net_pkt_append():1179
0x2000eb58/1    used    TDATA   net_ipv4_create_raw():37
0x2000ebe8/1    used    TDATA   net_pkt_append():1179
0x2000ec78/1    used    TDATA   net_ipv4_create_raw():37
0x2000ed08/1    used    TDATA   net_pkt_append():1179
0x2000ed98/1    used    TDATA   net_ipv4_create_raw():37
0x2000ef48/1    used    TDATA   net_ipv4_create_raw():37
0x2000eeb8/1    used    TDATA   net_ipv4_create_raw():37
0x20009110/1    used    TX      zsock_sendto():236
0x20008b70/1    used    TX      zsock_sendto():236
0x2000efd8/1    used    TDATA   net_pkt_append():1179
0x200090d4/1    used    TX      zsock_sendto():236
0x20009098/1    used    TX      zsock_sendto():236
0x2000905c/1    used    TX      zsock_sendto():236
0x20009020/1    used    TX      zsock_sendto():236
0x20008fe4/1    used    TX      zsock_sendto():236
0x20008fa8/1    used    TX      zsock_sendto():236
0x20008f6c/1    used    TX      zsock_sendto():236
[... 126 lines cut ...]
0x20012158/1    used    TDATA   net_ipv4_create_raw():37
0x200121e8/1    used    TDATA   net_pkt_append():1179
0x20012278/1    used    TDATA   net_ipv4_create_raw():37
0x20012308/1    used    TDATA   net_pkt_append():1179
0x20012398/1    used    TDATA   net_ipv4_create_raw():37
0x200124b8/1    used    TDATA   net_ipv4_create_raw():37
0x2000950c      free    TX      zsock_sendto():236 -> zsock_sendto():274
0x20011df8      free    TDATA   net_ipv4_create_raw():37 -> zsock_sendto():274
0x2000770c      free    RX      eth_rx():389 -> packet_received():2296
0x200095d8      free    RDATA   eth_rx():432 -> packet_received():2296
0x200076d0      free    RX      eth_rx():389 -> packet_received():2296
0x20009a58      free    RDATA   eth_rx():432 -> packet_received():2296
shell> 
shell> 
shell> net mem
Fragment length 128 bytes
Network buffer pools:
Address         Size    Count   Avail   Name
0x20014be0      7680    128     128     RX
0x20014bfc      7680    128     63      TX
0x20014c78      18432   128     128     RX DATA (rx_bufs)
0x20014ca0      18432   128     1       TX DATA (tx_bufs)
shell> 

```

pfalcon, 2017-10-16:
Update form me: I tried to look into this issue on weekend. I don't have 2 frdm_k64f anyway, so I started with qemu as usual. But I immediately hit #4022 and instead made another attempt at it, which was successful and fix pending a merge. Over that time, I ran it probably about a hundred times, and once I indeed hit a case of TX buffer being exhausted (in all other cases, it was RX exhaustion, per #4022).

Anyway, with #4022 approaching a solution, I should be able to look closer into this.

pfalcon, 2017-10-16:
Also, https://github.com/zephyrproject-rtos/zephyr/issues/4353 . People say that echo_server/echo_client don't work at all (just like imprinted in my memory too).

pfalcon, 2017-10-26:
I didn't have much chance to look further into this specifically, and wasn't able to reproduce it under QEMU anymore, so I guess I need to move to frdm_k64f hardware.

Here's some idea though (based on my looking at various stuff in the meantime). sockets don't do cheating and define custom per-context TX buffer pool (like seems to be pretty common with other stuff, I especially love to see comments like that: https://github.com/zephyrproject-rtos/zephyr/pull/980/files#diff-7a0cbc8ae7cc03faf53102d6d9323715R50).

Neither it does any TX flow control - once it finishes preparing and sending one packet, it will go allocate next one. So, it could exhaust free TX packets, and then it just exposes some deadlock we have in the stack. Finger in the sky, and ARP record got expired, so a next packet could not be sent. So, ARP code goes to allocate a pkt for its stuff, but there're no free packets, so it all nicely deadlocks.

There're definitely deadlocks like that (or we wouldn't have comments like quoted above), actually, few months ago we had more of them, and some definitely remains. ARP is just one plausible scenario, which need to be investigated.

gurucharanx, 2017-11-21:
I have found that the communication between sockets/echo and echo_client works fine if I apply  
[sockets_echo.txt](https://github.com/zephyrproject-rtos/zephyr/files/1491059/sockets_echo.txt)
the attached file changes to prj.conf file of sockets/echo.

But the communication is very slow as compared the communication between echo_server and echo_client applications.

I have verified by adding these changes on 
commit 25737d996349fb53ad8fe1716da049dce9fc585c

nashif, 2018-02-20:
no further comments, please reopen if issue still exists.

closed 2018-02-20 by nashif