zephyrproject-rtos/zephyr #4217: samples:net:sockets:echo_async : unexpected communication blocks between client and server after few packets transmission

gurucharanx, 2017-10-06:
This is a negative test for verifying the communication between the echo_async sample application as a TCP server and echo_client as a TCP client 

In this test there are three issues
1) As described in https://github.com/zephyrproject-rtos/zephyr/issues/4213
where an Ipv6 socket gets connected even though the configuration setting(CONFIG_NET_IPV6=n) is unset

2) As described in https://github.com/zephyrproject-rtos/zephyr/issues/4216
 The data packet transmission blocks after few packets transmission 
3) The logs clearly shows that the socket has been connected in IPv6 domain.But the logs at the client end shows that it can't connect and Init TCP in IPv6 

The configuration settings for Server and client are as below

At Server:
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_APP_SETTINGS=y
CONFIG_NET_APP_NEED_IPV4=y
CONFIG_NET_APP_NEED_IPV6=y
CONFIG_NET_APP_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_APP_PEER_IPV4_ADDR="192.0.2.2"
CONFIG_NET_APP_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_APP_PEER_IPV6_ADDR="2001:db8::2"

At Client:
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_UDP=n
CONFIG_NET_TCP=y
CONFIG_NET_APP_SETTINGS=y
CONFIG_NET_APP_CLIENT=y
CONFIG_NET_APP_NEED_IPV4=y
CONFIG_NET_APP_NEED_IPV6=y
CONFIG_NET_APP_MY_IPV6_ADDR="2001:db8::2"
CONFIG_NET_APP_PEER_IPV6_ADDR="2001:db8::1"
CONFIG_NET_APP_MY_IPV4_ADDR="192.0.2.2"
CONFIG_NET_APP_PEER_IPV4_ADDR="192.0.2.1"

The console o/p after the applications running at the server and client are as below

At server:
[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:28:f0:e2
�[dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:79:74:2a
[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
Listening on port 4242...
[dev/eth_mcux] [INF] eth_mcux_phy_event: Link do[dev/eth_mcux] [INF] ethvent: EnableConnection from 2001:db8::2 fd=536882888

At Client:
shell> [dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:a3:37:d6
shell> [dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[echo-client] [INF] init_app: Run echo client
[echo-client] [DBG] tcp_received: (0x20002e70): Sent 607 bytes, received 54 bytes
[echo-client] [DBG] compare_tcp_data: (0x20002e70): Compared 54 bytes, all ok
[echo-client] [DBG] tcp_received: (0x20002e70): Sent 607 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x20002e70): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x20002e70): Sent 607 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x20002e70): Compared 128 bytes, all ok
[echo-client] [ERR] connect_tcp: Cannot connect TCP (-60)
[echo-client] [ERR] start_tcp: Cannot init IPv6 TCP client (-60)
[echo-client] [INF] main: Stopping...

The same happens with the echo_async sample application too.

jukkar, 2017-10-12:
I tried to reproduce this but failed as the echo_async sample does not even compile with given conf file.
Could you provide exact instructions how you are compiling and running the samples?

gurucharanx, 2017-10-12:
@jukkar , I have used the same configuration file as in the respective folders expect adding/changing the above configurations in the respective files.

jukkar, 2017-10-16:
This is most probably the same issue as #4216 so lets concentrate the fixing efforts there.

laperie, 2018-03-05:
@gurucharanx Since the #4216 is fixed, can you please re-test if this one is fixed too as @jukkar  suggested?

kumarvikash1, 2018-03-24:
@tarunkum , please re-test this issue and update results.

tarunkum, 2018-03-27:
@jukkar 

I tried to communicate the two FDM boards with different sample application. 

In one target I have flashed the "Zephyr/samples/net/sockets/echo_async" & in other target I have flashed the "Zephyr/samples/net/echo_client".

After flashing I did not find any issues. Please find the below logs from two zephyr target devices.

window 1

Zephyr/samples/net/sockets/echo_async

***** BOOTING ZEPHYR OS v1.11.0-rc3 - BUILD: Mar 27 2018 05:55:44 *****
Asynchronous TCP echo server waits for connections on port 4242...
Connection #0 from 2001:db8::2 fd=536877480
Connection #1 from 192.0.2.2 fd=536877592

window 2

Zephyr/samples/net/echo_client

[echo-client] [DBG] tcp_received: (0x2000205c): IPv6: Sent 220 bytes, received 38 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 38 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv4: Sent 295 bytes, received 74 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 74 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv4: Sent 295 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv6: Sent 227 bytes, received 54 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 54 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv4: Sent 295 bytes, received 93 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 93 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv6: Sent 227 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv6: Sent 227 bytes, received 45 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 45 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv4: Sent 536 bytes, received 74 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 74 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv6: Sent 276 bytes, received 54 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 54 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv4: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv6: Sent 276 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv4: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv6: Sent 276 bytes, received 94 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 94 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv4: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv4: Sent 536 bytes, received 78 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 78 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv6: Sent 536 bytes, received 54 bytes
[echo-client] [DBG] compare_tcp_data: (0x2000205c): Compared 54 bytes, all ok
[echo-client] [DBG] tcp_received: (0x2000205c): IPv6: Sent 536 bytes, received 128 bytes

tarunkum, 2018-05-03:
@jukkar @kumarvikash1 

After re-validating I am not able to find any issues. 

I have again taken the below logs & this issue can be closed.

[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 45 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 54 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 54 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 98 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 98 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 536 bytes, received 74 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 74 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 536 bytes, received 78 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 78 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 54 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 54 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 98 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 98 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 436 bytes, received 74 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 74 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 436 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 436 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 436 bytes, received 106 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 106 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 502 bytes, received 54 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 54 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 502 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 502 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 502 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 502 bytes, received 64 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 64 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 384 bytes, received 74 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 74 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 384 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 384 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 384 bytes, received 54 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 54 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 348 bytes, received 54 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 54 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 348 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 348 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 348 bytes, received 38 bytes
[... cut ...]
[echo-client] [DBG] tcp_received: (0x200025d0): IPv6: Sent 536 bytes, received 98 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 98 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 536 bytes, received 74 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 74 bytes, all ok
[echo-client] [DBG] tcp_received: (0x200025d0): IPv4: Sent 536 bytes, received 128 bytes
[echo-client] [DBG] compare_tcp_data: (0x200025d0): Compared 128 bytes, all ok

closed 2018-05-03 by jukkar