zephyrproject-rtos/zephyr #4258: samples/net/zoap_server : unable to communicate between zoap client and server

gurucharanx, 2017-10-11:
I was trying to make a communication between the zoap_server and zoap_client but it was not communicating properly but giving as the Invalid UDP data received at the zoap_client application and there is no communication happening in between.

**The console output at the client side :**

[dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:ca:fd:b6
[dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:e1:13:4a
[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
Invalid UDP data received

**The console output at the server side:**

[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:97:16:f6
[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[dev/eth_mcux] [INF] eth_mcux_phy_event: Link down

rbtchc, 2017-10-16:
@gurucharanx I suggest that you enable the net debug info and test it again. So we could have more information.
I've been working on few patches (#4346) recently for coap_client and coap_server (on master branch) which might solve your issue.  However, they are not yet merged into the upstream.
I've tried to flash coap_client and coap_server into two NXP freedom k64f boards. They can talk to each other without problems.

jukkar, 2017-10-16:
> I've tried to flash coap_client and coap_server into two NXP freedom k64f boards. They can talk to each other without problems.

Thanks Robert for the info. Was this with your #4346 patch applied?

rbtchc, 2017-10-16:
@jukkar I tested w/ #4346 applied. I didn't test against the original implementation though (zoap_server/zoap_client).

gurucharanx, 2017-10-26:
I was expecting the applications of ZOAP server and ZOAP client to be communicated. Will it be able to communicate?

rbtchc, 2017-10-26:
@gurucharanx I tested against the **master** branch and retested again today.

> commit fb7f6cfa972d6eb122f37f195382f54148155725
> Author: Ding Tao <miyatsu@qq.com>
> Date:   Wed Oct 25 19:59:52 2017 +0800

They can talk to each other on NXP FRDM K64 board without any problem.
I did not test zoap_server / zoap_client. For they are renamed as coap_server/coap_client because zoap APIs are going to be deprecated.

gurucharanx, 2017-11-21:
I dont see the zoap_server and zoap_client applications in zephyr base of
commit: 25737d99
Author: Vinayak Kariappa Chettimada <vich@nordicsemi.no>
Date: Mon Nov 20 14:15:46 2017 +0100

@rbtchc  Have we merged these applications into coap_server/coap_client. May I know the reason behind of testing only coap_server/coap_client but not the zoap_server/zoap_client.

rveerama1, 2017-11-21:
Hi,
zoap_server/zoap_client are based on ZOAP library. Now ZOAP lib is 
deprecated, new CoAP is ZOAP + handles multi fragment (longer network 
packets).
So I renamed zoap_server/zoap_client to coap_server/coap_client to use 
new CoAP library api's and samples are same as zoap_server/zoap_client 
as purpose wise.

- Ravi.

gurucharanx, 2017-11-23:
Verified the communication between the coap_server and coap_client on frdm_k64f board and find it as working fine as explained in README.rst file of coap_client. So closing this issue.

The test was verified on commit id: b678895a6dadff7a10cb5ff6927ae74cc49fbae0

closed 2017-11-23 by gurucharanx