zephyrproject-rtos/zephyr #4276: Add check to flash_write to validate offset/length based on write block size

galak, 2017-10-11:
Now that we have flash_get_write_block_size() we should look to check offset & length are modulo the block size on flash_write() calls.

nvlsianpu, 2017-10-12:
Every driver in zephyr implements such check internally, along with memory boundaries check. This have advantage of that it isn't inline a function as write API is.
```C
static inline int flash_write(struct device *dev, off_t offset,
			      const void *data, size_t len);
```
Of course I see little gain (coherence+easier maintenance+easier addition of a new driver) in putting this check into flash_write API.

@carles FYI

nvlsianpu, 2020-11-13:
Each driver check this on it's own. Closed as Wont Do.

closed 2024-02-22 by nashif