zephyrproject-rtos/zephyr #4277: write a proper test for flash

galak, 2017-10-11:
We should write a proper test of the flash driver interface.  We can take a look at samples/drivers/soc_flash_nrf5 as a starting point.  This should check various things like mis-aligned erases/writes, validate write protection being set, etc.

MaureenHelm, 2019-09-04:
Is it possible to rework `tests/drivers/flash_simulator`

nvlsianpu, 2019-09-05:
> Is it possible to rework tests/drivers/flash_simulator

Flash simulator tests are testing some functionalities which are expected from flash_sim and not from regular flash driver. Although some code form it might be useful. @kapi-no What is your opinion?

`samples/drivers/soc_flash_nrf5` might be a good base for flash driver test suite.

kapi-no, 2019-09-05:
@nvlsianpu, I agree with you. ```tests/drivers/flash_simulator``` are intended to test flash_sim implementation to check if it's a good enough approximation of the real flash behaviour.

The driver is mainly used to test libraries that depend on the flash driver (e.g. NVS file system, Settings), but I don't mind if it is used for other purposes.

kartben, 2023-12-14:
@de-nordic is it still a valid issue?