zephyrproject-rtos/zephyr #4289: samples/mpu/mem_domain_apis_test is broken

SebastianBoe, 2017-10-12:
Hi @chunlinhan , git shows you as the author of this sample.

There seem to be several issues with the sample "mpu/mem_domain_apis_test".

It's not clear which boards it should be compatible with, I'm not that great at reading these yaml fields,
but I tried with bbc_microbit and qemu_cortex_m3, and neither would build.

Usually I run sanitycheck on a sample to find out what boards it supports, but in this case
sanitycheck would not build it against any board. This is unfortunate, as it will be doomed to
suffer from bit rot.

AnaZ implied that the filter was incorrect, that it should be CONFIG_USERSPACE, I'm not sure how the filter field works.

<•anaZ> Anas Nashif sebo_:
1:47 PM --- a/samples/mpu/mem_domain_apis_test/sample.yaml
1:47 PM +++ b/samples/mpu/mem_domain_apis_test/sample.yaml
1:47 PM @@ -5,5 +5,5 @@ tests:
1:47 PM  -   test:
1:47 PM          arch_whitelist: arm
1:48 PM          build_only: true
1:48 PM -        filter: USERSPACE
1:48 PM +        filter: CONFIG_USERSPACE
1:48 PM          tags: samples
1:48 PM sebo_:  i would open an issue and assign to Chunlin
1:48 PM it fails on nrf boards

`sample:
    description: TBD
    name: TBD
tests:
-   test:
        arch_whitelist: arm
        build_only: true
        filter: USERSPACE
        tags: samples
`

To be specific
- This sample should build against the boards it is expected to work with.
- This sample should (I assume) be picked up by sanitycheck to prevent bit rot

SebastianBoe, 2017-10-12:
mpu_test might be in the same boat.

andrewboie, 2017-10-12:
we need to move this test under tests/kernel/mem_protect/ and enable runtime success/failure reporting.

I doubt it works on any ARM target right now because CONFIG_USERSPACE is not fully enabled on ARM, although we should still be able to build it.

@AdithyaBaglody has been working on porting the memory domain APIs to x86, where we should be able to test end-to-end, see #1516

andrewboie, 2017-10-12:
mpu_test also needs to be moved and turned into a real test case

chunlinhan, 2017-10-16:
Hi @SebastianBoe  , sorry for my late reply.
I think this sample could be built and run on frdm_k64f or stm32f4_disco that supports MPU.
Both bbc_microbit and qemu_cortex_m3 do not support MPU yet, so you got build errors for
these two boards.

Regarding to sanitycheck issue, I will modify the yaml file to specify the boards to make sure
this sample will be built every time we run sanitycheck script.

PR #4425 (2017-10-19): samples: fix mem_domain_apis_test doesn't be built during sanitycheck
Use platform_whitelist to specify supported boards to make sure
mem_domain_apis_test sample will be built during sanitycheck.

Fixes #4289
closed 2017-10-19 by nashif