zephyrproject-rtos/zephyr #4292: net: tcp.c: prepare_segment() may unrightly unref packet in case of error

pfalcon, 2017-10-12:
Looking at the prepare_segment() function (https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/tcp.c#L328), it may be called either with a pre-existing packet, or create one itself. However, in multiple places, in case of error, it may call net_pkt_unref() on the pre-existing packet. This violates our policy that in the case of error, it's caller responsibility to clean up the packet (because e.g. it may want to retry it, or queue up for later retrying).

pfalcon, 2017-10-12:
Well, not only that. It has code like: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/tcp.c#L346 , which means that in case of early error return (e.g. https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/tcp.c#L384), the original fragment chain of incoming packet will be lost and leaked.

jukkar, 2017-10-13:
This fixes the double free issue described in #4216 
[net/pkt] [ERR] net_pkt_unref_debug: *** ERROR *** pkt 0x2000950c is freed already by prepare_segment():386 (zsock_sendto():274)

Unfortunately the buffer leak is still there.

closed 2017-10-14 by andrewboie

pfalcon, 2017-10-14:
> Unfortunately the buffer leak is still there.

Potentially, because https://github.com/zephyrproject-rtos/zephyr/issues/4292#issuecomment-336139174 wasn't addressed. I'm all for "one issue per ticket" (also, "one commit per pull request"), so let's have that as https://github.com/zephyrproject-rtos/zephyr/issues/4323 .

PR #4308 (2017-10-14): net: tcp: Do not unref caller allocated net_pkt
If the caller has passed net_pkt to prepare_segment(), then
it is caller responsibility to unref it in a case of error.

Fixes #4292
commit 059959c8 (2017-10-14): net: tcp: Do not unref caller allocated net_pkt

If the caller has passed net_pkt to prepare_segment(), then
it is caller responsibility to unref it in a case of error.

Fixes #4292