zephyrproject-rtos/zephyr #4295: Error flashing board STM32373C-EVAL

jffernandez, 2017-10-12:
When I try to flash the blinky sample on that board, I'm getting this output:
`make BOARD=stm32373c_eval flash`
`...`
`Flashing stm32373c_eval`
`Flashing Target Device`
`GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-00113-g0f83948 (2017-01-24-20:42)`
`Licensed under GNU GPL v2`
`For bug reports, read`
`	http://openocd.org/doc/doxygen/bugs.html`
`adapter speed: 1000 kHz`
`adapter_nsrst_delay: 100`
`Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD`
`none separate`
`srst_only separate srst_nogate srst_open_drain connect_deassert_srst`
`Info : Unable to match requested speed 1000 kHz, using 950 kHz`
`Info : Unable to match requested speed 1000 kHz, using 950 kHz`
`Info : clock speed 950 kHz`
`Error: open failed`
`in procedure 'init'`
`in procedure 'ocd_bouncer'`

`make[2]: *** [flash] Error 1`
`make[1]: *** [sub-make] Error 2`
`make: *** [flash] Error`

I traced the problem and seems that an OpenOCD configuration for ST-LINK/V2-1 is being used, but this board has an ST-LINK/V2 (as stated in the board docs in Zephyr Project and ST's website)

erwango, 2017-10-12:
Hi Jose,

You can have a try changing stlink config in openocd:
zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/board/st_nucleo_f3.cfg
-source [find interface/stlink-v2-1.cfg]
+source [find interface/stlink-v2.cfg]

And check it allows to flash.
Then about the board, is that V2 for all revisions for that board?
If yes, we can update Zephyr config. But usually, this varies between revisions (recent boards getting V2.1)

jffernandez, 2017-10-12:
Hi, thank you for the coments!

I changed the config of OpenOCD in boards/arm/stm32373c_eval/support/openocd.cfg
Now it works well (I will submit a patch)

I purchased the board a month ago at Mouser, its REV.B, but I'm pretty sure that the ST-LINK type not changed across versions, it's not documented anywhere.
Newer boards (e.g. NUCLEO-...) include the ST-LINK/V2-1, but previous models keep ST-LINK/V2.

This is what I can find, if somebody with that board can confirm, please!

Thank you!

jffernandez, 2017-10-13:
The solution of @erwango worked for me too.
If you can't modify your OpenOCD because are not using zephyr-sdk and that is installed on your system or whatever, you can try to modify boards/arm/stm32373c_eval/support/openocd.cfg like in my commit fenix-es/zephyr@80a1e9094320ae3003f6a25672b4054b755ede5a
This will be addressed in future releases with board revision as a Kconfig option.

closed 2017-10-13 by jffernandez