zephyrproject-rtos/zephyr #4301: checkpatch.pl false positives block PR merge

andrewboie, 2017-10-12:
If checkpatch.pl doesn't like a patch, and the error is a false-positive, the patch is blocked from being merged with no recourse for the developer.

Example, from #4299 

```
-:370: ERROR:COMPLEX_MACRO: Macros with complex values should be enclosed in parentheses
#370: FILE: kernel/include/syscall_handler.h:344:
+#define _SYSCALL_HANDLER0_SIMPLE_VOID(name_) \
+	_SYSCALL_HANDLER0(name_) { \
+		_impl_ ## name_(); \
+		return 0; \
+	}
```

This error is bogus, adding parens would break the string concatenation being done here.

nashif, 2017-10-12:
well, in this case just add a comment that those issues are bogus and have someone with override power verify it is a false positive and merge the PR.

andrewboie, 2017-10-12:
how many people have override power?
I can merge patches yet I don't think I can even do this.

nashif, 2017-10-12:
all members of the tsc and owners of the project should have this permission. So you should mention the tsc and ask for an override.

nashif, 2017-10-17:
override possible by anyone with admin access.

closed 2017-10-17 by nashif