zephyrproject-rtos/zephyr #4310: unable to flash quark_se_c1000_devboard

akhileshupadhyay, 2017-10-13:
After build passes, at time of flashing, board flashing script is not able to flash on quark_se_c1000_devboard. It gets _invalid command name "targets 1"._

looks that "_targets 1_" which is passed in the file "_ZEPHYR_BASE/boards/x86/quark_se_c1000_devboard/Makefile.board_"
introduces the issue.

================================
Steps to reproduce:
1. connect quark_se_c1000_devboard to host Linux machine
2. cd $ZEPHYR_BASE
3. source zephyr-env.sh
4. cd $ZEPHYR_BASE/samples/hello_world
5. make BOARD=quark_se_c1000_devboard flash

=================================

Below logs appear for flash failure:

AR      libapplication.a
  LINK    zephyr.lnk
  SIDT    staticIdt.o
  BIN     zephyr.bin
Flashing quark_se_c1000_devboard
Open On-Chip Debugger 0.9.0-dirty (2017-05-16-18:46)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
trst_only separate trst_push_pull
jtag_ntrst_delay: 300
Info : clock speed 1000 kHz
Info : JTAG tap: quark_se.cltap tap/device found: 0x0e765013 (mfg: 0x009, part: 0xe765, ver: 0x0)
Enabling arc core tap
Info : JTAG tap: quark_se.arc-em enabled
Enabling quark core tap
Info : JTAG tap: quark_se.quark enabled
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0  quark_se.quark     quark_se   little quark_se.quark     running
 1* quark_se.arc-em    arc32      little quark_se.arc-em    halted
invalid command name "targets 1"
Error: Command '['zephyr/sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd', '-s', 'zephyr/sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts', '-f', 'ZEPHYR_BASE/boards/x86/quark_se_c1000_devboard/support/openocd.cfg', '-c', 'init', '-c', 'targets', '-c', '"targets 1"', '-c', 'reset halt', '-c', 'load_image ZEPHYR_BASE/samples/hello_world/outdir/quark_se_c1000_devboard/zephyr.bin 0x40030000', '-c', 'reset halt', '-c', 'verify_image ZEPHYR_BASE/samples/hello_world/outdir/quark_se_c1000_devboard/zephyr.bin 0x40030000', '-c', 'reset run', '-c', 'shutdown']' returned non-zero exit status 1
Re-run with KBUILD_VERBOSE=1 for a stack trace.
ZEPHYR_BASE/Makefile:1379: recipe for target 'flash' failed
make[2]: *** [flash] Error 1
make[2]: Leaving directory 'ZEPHYR_BASE/samples/hello_world/outdir/quark_se_c1000_devboard'
Makefile:178: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
make[1]: Leaving directory 'ZEPHYR_BASE'
ZEPHYR_BASE/Makefile.inc:88: recipe for target 'flash' failed
make: *** [flash] Error 2

The commit ID that introduced this issue: bee6f9e73e6d494c0c67764d67d2ac30bc820c23

The commit ID where it works:  61aa0e06f5ab2ff0c322818853d11363a7ddd167 

Work around:

1. In the file "ZEPHYR_BASE/boards/x86/quark_se_c1000_devboard/Makefile.board"
replace OPENOCD_PRE_CMD = "targets 1"     with  OPENOCD_PRE_CMD = targets 1
2. connect quark_se_c1000_devboard to host Linux machine
3. cd $ZEPHYR_BASE
4. source zephyr-env.sh
5. cd $ZEPHYR_BASE/samples/hello_world
6. make BOARD=quark_se_c1000_devboard flash

Flashing should be successful.

nashif, 2017-10-13:
i have similar issues with arduino 101 and DFU flashing

mbolivar, 2017-10-13:
Please try https://github.com/mbolivar/zephyr/tree/issue-4310

mbolivar, 2017-10-13:
Actually, first attempt was buggy, please re-fetch (I've force-pushed) and try again. Sorry about that.

mbolivar, 2017-10-13:
NB Arduino 101 via DFU won't be fixed by that branch. I wasn't able to test this configuration (only have a DFUSE board on hand); can anyone pastebin the output from:

```
KBUILD_VERBOSE=1 ZEPHYR_FLASH_OVER_DFU=y BOARD=arduino_101 make -j8 flash
```

From samples/hello_world?

mbolivar, 2017-10-13:
I've identified an issue with dfu-util flashing and pushed https://github.com/mbolivar/zephyr/commit/687202c68870776bcb1269ad8cc1b62d8d49faf3, which is hopefully a fix. Please let me know how it goes.

mbolivar, 2017-10-13:
Update: I've added another couple of dfu-util fixes. This is still working on 96b_carbon for me, but it should behave more exactly like the shell script now.

PR #4316 (2017-10-13): Fixes for Python flashing
Fixes #4310
closed 2017-10-13 by nashif