zephyrproject-rtos/zephyr #4323: net: tcp.c: prepare_segment() may leak fragments in case of error

pfalcon, 2017-10-14:
Continuing from https://github.com/zephyrproject-rtos/zephyr/issues/4292#issuecomment-336139174 :

Well, not only that. It [prepare_segment()] has code like: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/tcp.c#L346 , which means that in case of early error return (e.g. https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/tcp.c#L384), the original fragment chain of incoming packet will be lost and leaked.

jukkar, 2017-10-27:
Is there still some issue with this, I thought this was fixed by #4292 or is the bug description wrong.

pfalcon, 2017-10-27:
No, this wasn't fixed by #4292, this is a separate issue, and the title and description are correct.

I'd say, this waits triaging by the maintainers, to confirm the issue and priority of "high".

linkmeyer, 2017-11-13:
@jukkar , @tbursztyka , what's the status?

jukkar, 2017-11-14:
The #4907 looks like a perfect match that would fix this.

commit c6407659 (2017-11-15): net: tcp: Add the frag back to caller allocated net_pkt

The original fragment chain of incoming packet will be lost and leaked
in case of early error, add frag back to packet and
let the caller do unref.

Fixes #4323
closed 2017-11-15 by jukkar