zephyrproject-rtos/zephyr #4324: samples/net/http_client: error in detecting and processing the message received

gurucharanx, 2017-10-14:
The http_client sample application is unable to processing the messages received from the Apache server and after few packets received it crashes the zephyr board
The test was done for IPv4

**Steps followed:**
**At local Linux machine**
This will act as an http server
1) Install the Apache server 2.4 in your local machine
2) $ cd /etc/httpd/conf/
3) Add the below setting in "httpd.conf" file 
Listen 8000
4) copy the index.html 
[index.txt](https://github.com/zephyrproject-rtos/zephyr/files/1384511/index.txt)] and 
post_test.php [post_test.txt](https://github.com/zephyrproject-rtos/zephyr/files/1384510/post_test.txt) files  in /var/www/html/
Note: please rename the attached files
5) restart the http server 
$systemctrl restart httpd.service
6) verify the status of http server by issuing the below
$service httpd status
7) If it is failing to start because of SElinux issues then issue the below command and restart as in step 4
$setenforce 0

**At zephyr end**
1) goto your zephyr source 
$cd zephyr/samples/net/http_client
2) change the below configuration in the prj.cof file for IPv4 http_client.

*CONFIG_NET_IPV4=y*
*CONFIG_NET_IPV6=n*
*CONFIG_NET_DHCPV4=n*
*CONFIG_DNS_RESOLVER=n*

3) $make clean;make pristine && make BOARD=qemu_x86 run 
The console output I could see is as below

  LINK    zephyr.lnk
  SIDT    staticIdt.o
  GMMU    mmu_tables.o
  GDT     gdt.o
  BIN     zephyr.bin
To exit from QEMU enter: 'CTRL+a, x'
[QEMU] CPU: qemu32
qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin

shell> [http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP body: 171 bytes, expected: 171 bytes
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [ERR] do_sync_http_req: **Error detected during HTTP msg processing**
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP body: 144 bytes, expected: 144 bytes
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP body: 171 bytes, expected: 171 bytes
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [ERR] do_sync_http_req: **Error detected during HTTP msg processing**
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP body: 144 bytes, expected: 144 bytes
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP body: 171 bytes, expected: 171 bytes
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [ERR] do_sync_http_req: **Error detected during HTTP msg processing**
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP body: 144 bytes, expected: 144 bytes
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP body: 171 bytes, expected: 171 bytes
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
[http-client] [INF] do_sync_http_req: HTTP server response status: OK
.
.
.
.
.
.
.***** CPU Page Fault (error code 0x00000000)
Supervisor thread read address 0x00000004
PDE: 0x025 Present Read-only User
PTE: 0x000 Non-present Read-only Supervisor
Current thread ID = 0x00406300
Faulting segment:address = 0x0008:0x0000c45a
eax: 0x00000000, ebx: 0x00000000, ecx: 0x00000000, edx: 0x00000000
esi: 0x00000246, edi: 0xaaaaaaaa, ebp: 0x00419390, esp: 0x00419380
eflags: 0x6
Fatal fault in essential thread! Spinning...

pfl, 2017-12-11:
The error string "[http-client] [ERR] do_sync_http_req: Error detected during HTTP msg processing" is only present in some older version of Zephyr. Current upstream since October 22nd would print the error differently. What is the last commit your Zephyr version is using ('git show HEAD')?

Please retest with latest upstream (>= 1.10, preferably latest upstream) if you haven't done so already and report if the bug still persists. If it does, please provide logs as above.

gurucharanx, 2017-12-12:
Currently the board is not crashing but still observing issues with post request as shown below

[http-client] [INF] response: Received 383 bytes piece of data
[http-client] [INF] response: HTTP server response status: Not Found
[http-client] [INF] response: HTTP body: 211 bytes, expected: 211 bytes
[http-client] [INF] do_async_http_req: [11] Send /index.html
[http-client] [INF] response: Received 425 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 171 bytes, expected: 171 bytes
[http-client] [INF] do_async_http_req: [11] Send /
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] do_async_http_req: [11] Send /post_test.php
[http-client] [INF] response: Received 330 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 144 bytes, expected: 144 bytes
[http-client] [INF] do_async_http_req: [11] Send /big-file.html
[http-client] [INF] response: Received 383 bytes piece of data
[http-client] [INF] response: HTTP server response status: Not Found
[http-client] [INF] response: HTTP body: 211 bytes, expected: 211 bytes
[http-client] [INF] do_async_http_req: [12] Send /index.html
[http-client] [INF] response: Received 425 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 171 bytes, expected: 171 bytes
[http-client] [INF] do_async_http_req: [12] Send /
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] do_async_http_req: [12] Send /post_test.php
[http-client] [ERR] do_async_http_req: **Cannot send POST request (-60)**
[http-client] [INF] main: Done!
QEMU: Terminated

I have verified this on commit c84a283d6866087ebb95a76515604b7b8bd4455e

For this test I have changed the below configuration settings in prj.conf file
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DEBUG_HTTP=n
CONFIG_NET_APP_NEED_IPV6=n
CONFIG_NET_APP_NEED_IPV4=y

Out of 5 times only 1 time it got passed but all other 4 times it failed.
Also attached the wireshark failure logs. Please rename the attached file accordingly to open with wireshark.
[http_client.txt](https://github.com/zephyrproject-rtos/zephyr/files/1550464/http_client.txt)

pfl, 2017-12-12:
"[http-client] [ERR] do_async_http_req: Cannot send POST request (-60)" is due to -ETIMEDOUT being reached in response to k_sem_take(&ctx->req.wait, timeout) in http_client_send_req() in subsys/net/lib/http/http_client.c. 'timeout' seems to be APP_REQ_TIMEOUT, which is defined as K_SECONDS(10).

The log output and wireshark failure logs don't match, they are from a different run with the latter containing far more packets exchanged. So it is not known what packets are sent before the timeout is hit after 'Send /post_test.php'.

gurucharanx, 2017-12-12:
@pfl ,the logs and wireshark taken at the same-time. Since there were so many requests and responses exchanged I didn't attach all the logs here. I thought the wireshark logs will help to debug the issue further when the failure happen.

rveerama1, 2017-12-19:
Can you test with https://github.com/zephyrproject-rtos/zephyr/pull/5442 ?

laperie, 2018-02-13:
@gurucharanx  -- please test with latest upstream and report

kumarvikash1, 2018-03-24:
@tarunkum , please test with latest upstream.

tarunkum, 2018-03-27:
@kumarvikash1 , @jukkar 

I tired today to validate the http_client on Zephyr & server on local machine.

I am not able to find any issues on http_client application & taken logs on zephyr console.

window 1

Zephyr/samples/net/http_client

[http-client] [INF] response: HTTP body: 24 bytes, expected: 24 bytes
[http-client] [INF] do_async_http_req: [18] Send /big-file.html
[http-client] [INF] response: Received 553 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 90 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 1528 bytes, expected: 1528 bytes
[http-client] [INF] do_async_http_req: [19] Send /index.html
[http-client] [INF] response: Received 347 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 161 bytes, expected: 161 bytes
[http-client] [INF] do_async_http_req: [19] Send /
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] do_async_http_req: [19] Send /post_test.php
[http-client] [INF] response: Received 163 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 24 bytes, expected: 24 bytes
[http-client] [INF] do_async_http_req: [19] Send /big-file.html
[http-client] [INF] response: Received 553 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 90 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 1528 bytes, expected: 1528 bytes
[http-client] [INF] do_async_http_req: [20] Send /index.html
[http-client] [INF] response: Received 347 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 161 bytes, expected: 161 bytes
[http-client] [INF] do_async_http_req: [20] Send /
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] do_async_http_req: [20] Send /post_test.php
[http-client] [INF] response: Received 163 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 24 bytes, expected: 24 bytes
[http-client] [INF] do_async_http_req: [20] Send /big-file.html
[http-client] [INF] response: Received 553 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 536 bytes piece of data
[http-client] [INF] response: Received 90 bytes piece of data
[http-client] [INF] response: HTTP server response status: OK
[http-client] [INF] response: HTTP body: 1528 bytes, expected: 1528 bytes
[http-client] [INF] main: Done!

nashif, 2019-02-19:
http-client removed, closing

closed 2019-02-19 by nashif