zephyrproject-rtos/zephyr #4325: samples/net/http_client:  unable to send the proper http request to Apache server

gurucharanx, 2017-10-14:
The http_client sample application is unable to send a proper http request to the Apache server due to which the server replies with bad request
The test was done for IPv6
The steps are as followed in #4324 except the configuration setting change  as below
*CONFIG_NET_IPV4=n*
*CONFIG_NET_IPV6=y*
*CONFIG_NET_DHCPV4=n*
*CONFIG_DNS_RESOLVER=n*

The console output (at qemu) is as below

  SIDT    staticIdt.o
  GMMU    mmu_tables.o
  GDT     gdt.o
  BIN     zephyr.bin
To exit from QEMU enter: 'CTRL+a, x'
[QEMU] CPU: qemu32
qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin

shell> [http-client] [INF] do_sync_http_req: HTTP server response status: Bad Request
[http-client] [INF] do_sync_http_req: HTTP body: 347 bytes, expected: 347 bytes
[http-client] [INF] do_sync_http_req: HTTP server response status: 
[http-client] [ERR] do_sync_http_req: Cannot send OPTIONS request (-58)
[http-client] [INF] main: Done!

jukkar, 2017-10-16:
Could you try #4348 if it solves the issue?

jukkar, 2017-10-17:
After #4348 I am still seeing sometimes this timeout error
```
[http-client] [INF] do_sync_http_req: HTTP body: 24 bytes, expected: 24 bytes
[http-client] [ERR] do_sync_http_req: Cannot send GET request (-60)
```

PR #4348 (2017-10-17): samples: net: http_client: Disable OPTIONS method
As many HTTP servers do not support OPTIONS method, disable it
by default.

Fixes #4325
commit 6f41e3a1 (2017-10-17): samples: net: http_client: Disable OPTIONS method

As many HTTP servers do not support OPTIONS method, disable it
by default.

Fixes #4325
closed 2017-10-17 by nashif