zephyrproject-rtos/zephyr #4326: Port Zephyr to Cypress PSoC 6 MCU's

raveslave, 2017-10-14:
Based on ARM M4 & M0+   
http://www.cypress.com/products/32-bit-arm-cortex-m4-psoc-6

Since their IDE "PSoC Creator" is Windows only, having an alternative tool-chain would be much appreciated by the community.

raveslave, 2018-03-03:
Seems Cypress made an effort and now supports an alternative tool-chain that would make this more doable without the previous psoc creator dependency to configure the IC & peripherals.

http://www.cypress.com/products/modustoolbox-integrated-design-environment-ide

pfalcon, 2018-08-18:
There's ongoing work on adding PSoC6 support happening in the branch https://github.com/nazrchorn/zephyr/tree/add_psoc6 .

pfalcon, 2018-08-18:
Also the PR: https://github.com/zephyrproject-rtos/zephyr/pull/8408

raveslave, 2018-08-19:
Great to see work being done on this! Thanks Cypress!

Historically in PSoC Creator, there has been quite good maintenance on the driver libraries (PDL's and also the Verilog modules) with updates (based on bug-reports) coming every now & then. 

I saw that there is now a hot-copy here https://github.com/cypress/pdl which is great.
@galak is there a plan to use github as the source from now on as the source for patches and new PDL releases? if yes, what's the best/smoothest way to integrate these back into Zephyrs driver-tree?

carlescufi, 2018-10-04:
@galak Can we close this now that this is merged?

raveslave, 2018-10-04:
I've checked this out and right now there is only a hard-coded UART supported.   
Would be great with the full peripheral-set of the PSoC 6 as well as offering a more flexible pin-configuration.

carlescufi, 2018-10-04:
@raveslave thanks for chiming in. @nazrchorn @OlegKapshii do you have plans to improve and extend the port?

OlegKapshii, 2018-10-15:
Yes. We will extend it.

raveslave, 2018-11-09:
Any plans on when?

pfalcon, 2018-11-09:
@raveslave : All the time ;-). You can search pull requests for "psoc6" for the current work in progress. E.g. right now you can help with reviewing and testing https://github.com/zephyrproject-rtos/zephyr/pull/10601. (And without help with testing, it's all going to be very-very slow.)

Oh, and btw, I wouldn't close this ticket, given the community interest. (I.e. this ticket can be subscribed to by interested parties, and updates can be given here once in a while).

raveslave, 2018-11-09:
We'd be happy to do testing, but we need more things in place first
- peripherals: 
  * i2c, spi, i2s, sdio, pdm, usb, etc..
  * other psoc 6 specifics, i.e. the analog blocks
- pin-config & routing

@galak can you please look into this and re-open this issue.

pfalcon, 2018-11-09:
@raveslave: As you can see, the current thing which needs testing is IPM communication between the cores.

Reopening based on the community request, reassigning to me. I'll try to post updates to this, and we can open at least until 1.14 release I guess.

raveslave, 2019-02-09:
@pfalcon @OlegKapshii what's the current status & plans here?

PingiliV, 2019-04-11:
Hello, Iam new to zephyr and psoc6. Is there any document or link where i can check the supported features/ Drivers for psoc6 particularly. Also, specific way to check issues related to Psoc6?
Thanks in advance for your effort !
BR
Vinay

OlegKapshii, 2019-04-12:
Hello,
All available documentation is in boards/arm/cy8ckit_062_wifi_bt_m0 (m4) doc folders. There are also some links to Cypress PSoC6 resources. At the moment, there is a basic PSoC6 support added (cy8ckit_062_wifi_bt board), a device tree created, a sample project can be compiled and loaded into a PSoC6 board (hello_world e.g.).
There are two UART ports in the device tree: one for CM0 core, one - for CM4.
There are also two PRs, which are waiting for merging into master (Psoc6 add ipm #10601 and Psoc6 add protection #11777), that should add some more complex features support for PSoC6 in Zephyr.

galak, 2020-03-10:
Basic support for PSOC6 is in zephyr.  Additional driver support would need to be provided by community members that have an interest.

closed 2020-03-10 by galak