zephyrproject-rtos/zephyr #4327: NET_PKT_TX_SLAB_DEFINE, NET_PKT_DATA_POOL_DEFINE description and usage are confusing

pfalcon, 2017-10-14:
Reading https://www.zephyrproject.org/doc/api/networking.html#c.NET_PKT_TX_SLAB_DEFINE :

> Create a TX net_pkt slab that is used when sending user specified data to network.

I don't think this does good to describe what it does. The "user specified" modifier for "data" sounds confusing (is there specific meaning behind it?). On the other hand, there may be missing "custom" in "Create a *custom* TX net_pkt slab" (that's the idea, that this macro is used to create further, custom packet slabs beyond the system default, right)? Also, what about RX counterpart? Without it, docs look confusing/incomplete, so it would be good to mention why there're no custom RX slabs.

Going next: https://www.zephyrproject.org/doc/api/networking.html#c.NET_PKT_DATA_POOL_DEFINE

> Create a DATA net_buf pool that is used when sending user specified data to network.

Again, "user specified" for "data" is confusing. Rather, would be "custom" for "pool", like above. And again, why "sending"? What about "receiving".

Let's no stop here though and look inside subsys/net/ip/net_pkt.c, and compare with the macros above:

~~~
#define NET_PKT_TX_SLAB_DEFINE(name, count)                             \
        K_MEM_SLAB_DEFINE(name, sizeof(struct net_pkt), count, 4)
#define NET_PKT_DATA_POOL_DEFINE(name, count)                           \
        NET_BUF_POOL_DEFINE(name, count, CONFIG_NET_BUF_DATA_SIZE,      \
                            CONFIG_NET_BUF_USER_DATA_SIZE, NULL)
~~~

~~~
#define NET_PKT_RX_COUNT        CONFIG_NET_PKT_RX_COUNT
#define NET_PKT_TX_COUNT        CONFIG_NET_PKT_TX_COUNT
#define NET_BUF_RX_COUNT        CONFIG_NET_BUF_RX_COUNT
#define NET_BUF_TX_COUNT        CONFIG_NET_BUF_TX_COUNT
#define NET_BUF_DATA_LEN        CONFIG_NET_BUF_DATA_SIZE
#define NET_BUF_USER_DATA_LEN   CONFIG_NET_BUF_USER_DATA_SIZE

K_MEM_SLAB_DEFINE(rx_pkts, sizeof(struct net_pkt), NET_PKT_RX_COUNT, 4);
K_MEM_SLAB_DEFINE(tx_pkts, sizeof(struct net_pkt), NET_PKT_TX_COUNT, 4);

NET_BUF_POOL_DEFINE(rx_bufs, NET_BUF_RX_COUNT, NET_BUF_DATA_LEN,
                    NET_BUF_USER_DATA_LEN, NULL);

NET_BUF_POOL_DEFINE(tx_bufs, NET_BUF_TX_COUNT, NET_BUF_DATA_LEN,
                    NET_BUF_USER_DATA_LEN, NULL);
~~~

What can be seen is: both macros in include/net/net_pkt.h and stuff in subsys/net/ip/net_pkt.c use the same underlying slab/pool definitions, but this fact is not apparent and actually rather masked by extra defines redirection in net_pkt.c.

Suggestions: turn NET_PKT_TX_SLAB_DEFINE, NET_PKT_DATA_POOL_DEFINE into generic macros used both by the system (to define default slab-pools) and user to define custom ones. Specific steps:

1. Rename NET_PKT_TX_SLAB_DEFINE to NET_PKT_SLAB_DEFINE, as there's nothing TX specific in its definition. Provide deprecated NET_PKT_TX_SLAB_DEFINE macro for compatibility.
2. Get rid of as many CONFIG_* defines indirections in net_pkt.c as practical (not necessary all).
3. Convert net_pkt.c to use generic NET_PKT_SLAB_DEFINE and NET_PKT_DATA_POOL_DEFINE macros.
4. Make those macros have general, helpful descriptions. E.g. NET_PKT_SLAB_DEFINE could have:

"Create a net_pkt slab used to store network packet information (without data, NET_PKT_DATA_POOL_DEFINE should be used for that) . User can define additional custom TX net_pkt slabs (no custom RX slabs are supported)."

pfalcon, 2017-10-14:
This comes as a side result of investigating https://github.com/zephyrproject-rtos/zephyr/issues/4022 .

I will be happy to provide patches, just would like to get an ack from @jukkar and @dbkinder that it makes sense.

dbkinder, 2017-10-16:
I'll need you developers to haggle over the technical details of this issue.  I can help tweak the wording for clarity once the technical aspects are OK.

jukkar, 2017-10-17:
Just send patches if you want to change something.

pfalcon, 2017-10-19:
@jukkar : I start with the assumption that if something is done in a particular way, then there's a reason for it, and it's better discuss the changes before investing time into them. So, I assume the plan above looks ok, and will go for it.

PR #4443 (2017-10-23): net: pkt: Use standard macros to define system pkt slabs/pools
Previously net_pkt.h, defined macros NET_PKT_TX_SLAB_DEFINE,
NET_PKT_DATA_POOL_DEFINE, but advertised them as intended for
"user specified data". However, net_pkt.c effectively used the
same parameters for slabs/pools, but this wasn't obvious due
to extra config param redirection. So, make following changes:

1. Rename NET_PKT_TX_SLAB_DEFINE() to NET_PKT_SLAB_DEFINE()
as nothing in its definition is TX-specific.
2. Remove extra indirection for config params, and use
NET_PKT_SLAB_DEFINE and NET_PKT_DATA_POOL_DEFINE to define
system  [... cut]
commit 09b96736 (2017-10-23): net: pkt: Use standard macros to define system pkt slabs/pools

Previously net_pkt.h, defined macros NET_PKT_TX_SLAB_DEFINE,
NET_PKT_DATA_POOL_DEFINE, but advertised them as intended for
"user specified data". However, net_pkt.c effectively used the
same parameters for slabs/pools, but this wasn't obvious due
to extra config param redirection. So, make following changes:

1. Rename NET_PKT_TX_SLAB_DEFINE() to NET_PKT_SLAB_DEFINE()
as nothing in its definition is TX-specific.
2. Remove extra indirection for config params, and use
NET_PKT_SLAB_DEFINE and NET_PKT_DATA_POOL_DEFINE to define
system [... cut]
closed 2017-10-23 by jukkar