zephyrproject-rtos/zephyr #4347: net: BSD Sockets UDP sendto() impl broke tests/net/socket/udp/

pfalcon, 2017-10-16:
After 9ec60774f8cfda70fcdd3e39dc45df40eed3cb21, https://github.com/zephyrproject-rtos/zephyr/tree/master/tests/net/socket/udp is broken (it hangs on run). @askawu , can you please look if it is possible to fix it?

askawu, 2017-10-16:
@pfalcon, based on master, I run the command "./scripts/sanitycheck -s tests/net/udp/tests" and it passed without error. Can you provide more information about how you produce the error? Thanks.

pfalcon, 2017-10-16:
Yep, we don't really *run* any network tests in sanitycheck, because it requires special environment setup (loop-slip-tap.sh, etc.)

So, you'd need to run it manually as usual:
~~~
cd tests/net/udp/tests
make run
~~~

askawu, 2017-10-16:
Got it. I will try it.

jukkar, 2017-10-16:
> Yep, we don't really run any network tests in sanitycheck, because it requires special environment setup (loop-slip-tap.sh, etc.)

That is not correct, we have lot of networking tests that are actually run with sanitycheck. They just depend on having a test program that can handle feeding data back to itself.

pfalcon, 2017-10-16:
Right, we have few unit-style networking tests running with sanitycheck. Nothing which does "integration" level testing (i.e. being able to take the same test app which runs under qemu and run it with sanitycheck.)

askawu, 2017-10-16:
The test case "test_send_recv_2_sock" failed because net_context_sendto() is called but dest address is NULL. I will submit a patch to fix it.

Besides, I would like to check the behavior that sendto() and recvfrom() are used for a stream socket on a Linux machine and add test cases on Zephyr

PR #4370 (2017-10-17): net: sockets: Fix the error zsock_send() failed on the datagram socket
net_context_sendto() returns an error if dest address is NULL.

If dest address is available, net_conext_sendto() should be used.
Otherwise, net_context_send() should be used.

Fixes #4347
commit 63290411 (2017-10-17): net: sockets: Fix the error zsock_send() failed on the datagram socket

net_context_sendto() returns an error if dest address is NULL.

If dest address is available, net_conext_sendto() should be used.
Otherwise, net_context_send() should be used.

Fixes #4347
closed 2017-10-17 by nashif