zephyrproject-rtos/zephyr #4351: USB device is not listed after flash with Zephyr

dhavalpanchalispl, 2017-10-16:
OS: Ubuntu 16.04
Device: Arduino 101 (Intel Curie Board)
Zephyr SDK: 0.9.1

I have flashed both ARC and X86 cores of Ardunio 101 board as mentioned in this sensor example,
https://www.zephyrproject.org/doc/samples/sensor/bmm150/README.html
And it successfully uploaded image to Arduino 101 device. Here is the output for "make BOARD=arduino_101_sss flash",

  CHK     include/generated/generated_dts_board.conf
  Using /home/zephyr as source for kernel
  GEN     ./Makefile
  CHK     include/generated/version.h
  CHK     include/generated/generated_dts_board.h
  CHK     include/generated/syscall_list.h
  CHK     include/generated/syscall_dispatch.c
  CHK     misc/generated/configs.c
  CHK     include/generated/offsets.h
make[3]: 'isr_tables.o' is up to date.
Flashing arduino_101_sss
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 8087:0aba
Run-time device DFU version 0011
Claiming USB DFU Interface...
Setting Alternate Setting #7 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0011
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        18260 bytes
Download done.
state(2) = dfuIDLE, status(0) = No error condition is present
Done!

But **I can not find any USB device** listed in my Ubuntu system. To see output logs of board I tried command, sudo picocom /dev/ttyUSB0. But there is no ttyUSB0 device. Also cutecom (sudo apt-get install cutecom) does not show any USB device after flashing. 

Output of this command to check connected tty devices: "dmesg | grep tty",
[    0.000000] console [tty0] enabled
[    0.819096] 00:02: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A

I connected Arduino 101 device using USB power cable. I have not used JTAG debugger. Is it necessary to use JTAG debugger if I just want to see execution logs of my program? I want to see those sensor values in console as mentioned in this example, https://www.zephyrproject.org/doc/samples/sensor/bmm150/README.html. But I could not see because USB device does not show.

jfischer-phytec-iot, 2017-10-16:
Hi,

if you build the sample with default settings then you need a additional TTL-to-USB adapter. **Required Hardware and Software**: "If you wish to grab any data off the serial port, you will need a TTL-to-USB adapter".

https://www.zephyrproject.org/doc/boards/x86/arduino_101/doc/board.html

nashif, 2017-10-16:
as mentioned by @jfischer-phytec-iot , you need a dedicated cable for serial, Arduino 101 does not send serial output over the usb cable connected to power the device and flash it.

nashif, 2018-02-20:
closing, no further comments

closed 2018-02-20 by nashif