zephyrproject-rtos/zephyr #4353: VM-VM qemu networking example crashes often

SebastianBoe, 2017-10-16:
Hi,

I was trying to test the VM-VM setup that is documented in "Networking with QEMU",
but the example that was suggested, echo_server with echo_client, is only able to transmit
a handful of packets before either hanging or crashing with a page fault.

BOARD=qemu_x86

I tried doing a git bisect, but the symptoms kept changing (e.g. hanging instead of crashing). 
This must be a regression because v1.8 is able to transmit at least hundreds of packets, 
whereas master crashes/hangs after a handful.

Revision of master that was tested:
commit a2af43e27070448223fec275636bb2af9a189400
Author: Robert Chou <robert.ch.chou@acer.com>
Date:   Mon Oct 16 16:23:51 2017 +0800

crash log:

> [QEMU] CPU: qemu32
> qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin
> 
> shell> [net/app] [INF] net_app_init: Initializing network
> [net/app] [INF] setup_ipv4: IPv4 address: 192.0.2.1
> [net/app] [INF] ipv6_event_handler: IPv6 address: fe80::200:5eff:fe00:537f
> [net/app] [INF] ipv6_event_handler: IPv6 address: fe80::200:5eff:fe00:537f
> [net/app] [INF] new_client: Connection from [2001:db8::2]:52478 (0x004033f0)
> [echo-server] [INF] build_reply_pkt: TCP IPv6 received 626 bytes
> [echo-server] [INF] build_reply_pkt: TCP IPv6 received 622 bytes
> [echo-server] [INF] build_reply_pkt: TCP IPv6 received 1080 bytes
> [echo-server] [INF] build_reply_pkt: TCP IPv6 received 917 bytes
> [net/app] [DBG] _net_app_accept_cb: (0x00403080): Already connected via context 0x004033f0
> [echo-server] [INF] build_reply_pkt: TCP IPv6 received 696 bytes
> [echo-server] [INF] build_reply_pkt: TCP IPv6 received 1014 bytes
> [echo-server] [INF] build_reply_pkt: TCP IPv6 received 1008 bytes
> [net/app] [DBG] net_app_send_pkt: (0x00403080): Cannot send to peer (-22)
> [echo-server] [ERR] tcp_received: Cannot send data to peer (-22)
> [echo-server] [INF] main: Stopping...
> ***** CPU Page Fault (error code 0x00000000)
> Supervisor thread read address 0x00000004
> PDE: 0x025 Present Read-only User
> PTE: 0x000 Non-present Read-only Supervisor
> Current thread ID = 0x00404730
> Faulting segment:address = 0x0008:0x000055b7
> eax: 0xffffffff, ebx: 0x00000000, ecx: 0x0000563a, edx: 0x004037b0
> esi: 0x00000000, edi: 0xaaaaaaaa, ebp: 0x0041e3bc, esp: 0x0041e3b8
> eflags: 0x286
> Fatal fault in thread 0x00404730! Aborting.

SebastianBoe, 2017-10-16:
This may share a root cause with https://github.com/zephyrproject-rtos/zephyr/issues/4188 , it is not clear how to reproduce 4188, but it should be clear how to reproduce this issue.

kumarvikash1, 2017-11-26:
What is the target release version ? Can we fix this issue in R-1.10 ?

nashif, 2017-11-28:
@SebastianBoe is this still an issue?

SebastianBoe, 2017-11-29:
Works perfectly!

Re-tested with a commit from master:

commit a8054d5936324e70dc9cdbfb4c37a89d5d08d17c
Author: Neil Armstrong <narmstrong@baylibre.com>
Date:   Tue Nov 28 16:40:40 2017 +0100

closed 2017-11-29 by SebastianBoe