zephyrproject-rtos/zephyr #4366: memory corruption in test_pipe_api

andrewboie, 2017-10-16:
There is some kind of horrible memory corruption going on in the pipe API test. I am not sure this is a problem in the test case, I think this may be a kernel bug.

To reproduce, apply the following patch:
```
diff --git a/tests/kernel/pipe/pipe_api/src/test_pipe_contexts.c b/tests/kernel/pipe/pipe_api/src/test_pipe_contexts.c
index cd665b532..c688aa468 100644
--- a/tests/kernel/pipe/pipe_api/src/test_pipe_contexts.c
+++ b/tests/kernel/pipe/pipe_api/src/test_pipe_contexts.c
@@ -30,8 +30,8 @@ K_PIPE_DEFINE(kpipe, PIPE_LEN, 4);
 static struct k_pipe pipe;
 
 static K_THREAD_STACK_DEFINE(tstack, STACK_SIZE);
-static struct k_thread tdata;
-static struct k_sem end_sema;
+struct k_thread tdata;
+struct k_sem end_sema;
 
 static void tpipe_put(struct k_pipe *ppipe)
 {
```

With tdata and end_sema in a different part of memory, I get a hard crash on any QEMU target I attempt with. You have to apply the patch to see this crash. q

This bug has been around for a while, it is not due to any recent activity, if I checkout the v1.9.0 tag and apply the above patch, I also get a crash.

What little investigation I did, it seems that the kernel's ready queue gets corrupted, and the system tries to swap in a bogus new value of _current, to something which is not a k_thread, and the stack pointer gets set to some garbage value (usually a NULL)

linkmeyer, 2017-11-06:
@andyross , what's your status in getting this bug fixed?  It's a high priority kernel bug which has been open for 20 days.

andyross, 2017-11-13:
Nothing too exotic as it happened: end_sema was just uninitialized.  And the corruption you got when it linked in after pipe was a whole lot crazier than when it was before the pipe, I don't have a good answer.

Note if you want to validate, the patch as submitted changes too many variables by using K_SEM_DEFINE (which also changes the linker order).  Just adding a k_sem_init() to the top of the various failing tests is enough to fix the issue too and works without change to the linker map.   But it's the wrong way to initialize a semaphore used in multple tests, so I didn't do that.

PR #4923 (2017-11-14): tests/kernel/pipe: fix uninitialized semaphore
The end_sema k_sem was only initialized on one of the several paths
that used it, leading to some crazy clobber-the-run-queue behavior
that was dependent on linkage order (see the linked bug) when end_sema
and the pipe object were made non-static..

Adding a k_sem_init() call fixes the corrupt issue, but really the
right thing is to use the DEFINE macro, so do that instead.  Note that
that the initializer changes the linkage order too (by putting the
semaphore in a separate segment), so... yeah, it's actually impossible
to prove that this patch in [... cut]
closed 2017-11-14 by andrewboie