zephyrproject-rtos/zephyr #4377: Sniffing traffic in a VM-VM qemu setup crashes with a segfault in the monitor application

SebastianBoe, 2017-10-17:
Hi,

it looks like using Makefile.ipstack's PCAP to sniff traffic in
a qemu VM-VM setup is broken.

I am observing that monitor_15_4 is segfaulting. I tried to find the root cause,
but was unable to.

# Environment to reproduce

To create traffic I am using echo_server and echo_client with qemu_x86.

NB: Make sure you use -j1 or else you will provoke another unrelated bug in Makefile.ipstack (ipstack should be doing this to get the dependencies correct "server: $(DOTCONFIG) | remove_pipes setup_pipes_dual_monitor start_monitor")

echo_server and echo_client will transmit some packets successfully and then probably crash
due to a presumable third unrelated bug (https://github.com/zephyrproject-rtos/zephyr/issues/4353).

The packets that were succesfully transmitted before the crash should be sniffed correctly, but
opening the pcap file shows no packets.

git revision:
commit a2af43e27070448223fec275636bb2af9a189400
Author: Robert Chou <robert.ch.chou@acer.com>
Date:   Mon Oct 16 16:23:51 2017 +0800

# Debugging done so far
I tried to find out what monitor_15_4 was doing when it should have been logging packets, so I ran it standalone instead of having the build system call it, then I observed that it would segfault.

> sebo@mach:~/net-tools$ /home/sebo/net-tools/monitor_15_4 /home/sebo/workspace_zephyr/misc/capture_file_kbuild.txt /tmp/ip-stack-server /tmp/ip-stack-client
> /home/sebo/net-tools/monitor_15_4 /home/sebo/workspace_zephyr/misc/capture_file_kbuild.txt /tmp/ip-stack-server /tmp/ip-stack-client
> Segmentation fault (core dumped)

I tried to find out why it segfaulted by building it with -g and running it under gdb, but the callstack didn't have any useful information.

```
sebo@mach:~/net-tools$ gdb --args /home/sebo/net-tools/monitor_15_4 /home/sebo/workspace_zephyr/misc/capture_file_kbuild.txt /tmp/ip-stack-server /tmp/ip-stack-client
gdb --args /home/sebo/net-tools/monitor_15_4 /home/sebo/workspace_zephyr/misc/capture_file_kbuild.txt /tmp/ip-stack-server /tmp/ip-stack-client
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/sebo/net-tools/monitor_15_4...done.
(gdb) c
c
The program is not being run.
(gdb) run
run
Starting program: /home/sebo/net-tools/monitor_15_4 /home/sebo/workspace_zephyr/misc/capture_file_kbuild.txt /tmp/ip-stack-server /tmp/ip-stack-client
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
  C-c C-c^C
Program received signal SIGINT, Interrupt.
0x00007ffff77f76f0 in __poll_nocancel () at ../sysdeps/unix/syscall-template.S:84
84	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) bt
bt
#0  0x00007ffff77f76f0 in __poll_nocancel () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007ffff7b1038c in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7b10712 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x0000000000401cba in main (argc=4, argv=0x7fffffffe498) at monitor_15_4.c:481

```

jukkar, 2017-10-17:
I just tried this like this:

```
cd samples/net/echo_server
echo "CONFIG_NET_TCP=y" >> prj_qemu_802154.conf
echo "CONFIG_NET_TEST=y" >> prj_qemu_802154.conf
make pristine && make CONF_FILE=prj_qemu_802154.conf PCAP=output.pcap server
```

```
cd samples/net/echo_client
echo "CONFIG_NET_TCP=y" >> prj_qemu_802154.conf
echo "CONFIG_NET_TEST=y" >> prj_qemu_802154.conf
make pristine && make CONF_FILE=prj_qemu_802154.conf client
```

The outcome of this was that:
* echo-client crashes, did not investigate yet why
* echo-server hangs after receiving some packets
* pcap file contains packets, but they are malformed, according to wireshark the end of the each packet contains some garbage bytes
* I did not see any issues or errors with monitor_15_4, my host system is Fedora 26

I will investigate this crashing more and will also send patch to those two config files that I needed to modify.

jukkar, 2017-10-18:
I have some status update to this one:

* For monitor_15_4 I fixed couple of uninitialized memory access issues, the PR is https://github.com/zephyrproject-rtos/net-tools/pull/14 In my linux host (Fedora 26) everything worked just fine without this PR thou.

* PR #4411 fixes pcap corruption issue
* PR #4410 enables the communication between two qemus

All is still not well after these fixes as echo-server crashes after receiving couple of packets. I am looking this one next.
Also client qemu is very fragile, most of the time it refuses to start at all. Dunno what is causing this one. If client qemu refuses to start, then I also need to restart the server qemu as otherwise the connectivity is not working as expected.

SebastianBoe, 2017-10-18:
Sounds like good progress. Will re-test when you are ready.

jukkar, 2017-10-19:
Two more PR are needed for everything to work:

* PR #4432 which fixes 6lo compression issue
* PR #4433 which fixes the crash issue in 802.15.4 reassembly code

After these total four PR #4410 #4411 #4432 and #4433 I managed to get qemu<->qemu communication to work. There are issues with qemu thou, it refuses to start sometimes. This seems to be related to monitor_15_4 and pipes. If I remove the PCAP option from "make server", then the system was working more reliably.

SebastianBoe, 2017-10-19:
Then hopefully https://github.com/zephyrproject-rtos/zephyr/issues/4353 has been resolved. Will re-test after the PR's have reached master.

PR #4432 (2017-10-20): Fix 6lo uncompression issue
6lo uncompression not handled properly when source address mode is SAM_00 and SAC is 1 (unspecified source address "::"). This is a special case, it should be handled without context based compression support.

Fixes #4377
PR #4433 (2017-10-24): net: ieee802154: Check reassembly return value properly
If IEEE 802.15.4 reassembly function ieee802154_reassemble() returns
anything other than NET_CONTINUE, then drop that packet. Earlier
it only dropped the packet if NET_DROP was returned but the reassembly
might also return NET_OK. In that case the pkt is freed already and
pkt->frags pointer is NULL. This caused NULL pointer access in L2 when
packet was received.

Fixes #4377
closed 2017-10-20 by jukkar