zephyrproject-rtos/zephyr #4401: tests/net/ipv6/test.yaml :--Cannot add multicast IPv6 address

rrahulkx, 2017-10-18:
This testcase is failed .It is not able to add Multicast address.

This testcase is failed at certain targets mentioned below :--

arduino101:x86
qemu-zephyr-x86
arduino101:x86
qemu-zephyr-x86
ma:x86
ma:x86

Toolchain: ZEPHYR_GCC_VARIANT=zephyr

Commit ID                                Branch          URL
---------------------------------------- --------------- -------------------
328f80fa073e4c56bc2ea2d5c230e5e6f4a4b5a6 master          https://github.com/zephyrproject-rtos/zephyr.git

The detail information is pasted below --

**Testcase :--**

tests/net/ipv6/testcase.yaml#test @ qc1000-01:x86 COMPLETION failed

Target Name  Remote ID               Type
-----------  ----------------------  ----
target            qc1000-01             ma

**Detailed description**

**Build Procedure :--**

configure passed

build passed: 'mkdir -p outdir-w28a-3ceh-quark_se_c1000_devboard; rm -f outdir-w28a-3ceh-quark_se_c1000_devboard/.config'
build passed: 'make -j -C zephyr/tests/net/ipv6/ EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:3ceh 
BOARD=quark_se_c1000_devboard ARCH=x86  
O=outdir-w28a-3ceh-quark_se_c1000_devboard initconfig'

build output: Using /zephyr/boards/x86/quark_se_c1000_devboard/quark_se_c1000_devboard_defconfig as base
build output: Merging zephyr/zephyr/tests/include/test.config
build output: Merging outdir-w28a-3ceh-quark_se_c1000_devboard/arc_target.conf
build output: Merging outdir-w28a-3ceh-quark_se_c1000_devboard/boot_config.conf
build output: #
build output: # configuration written to .config
build output: #
build passed: 'make -j -C zephyr/zephyr/tests/net/ipv6 EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" KCPPFLAGS=-DTC_RUNID=:3ceh BOARD=quark_se_c1000_devboard ARCH=x86  O=outdir-w28a-3ceh-quark_se_c1000_devboard'
build passed: 'mkdir -p outdir-w28a-3ceh-quark_se_c1000_ss_devboard; rm -f utdir-w28a-3ceh-quark_se_c1000_ss_devboard/.config'
build passed: 'make -j -C zephyr/zephyr/tests/booting/stub/ EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" KCPPFLAGS=-DTC_RUNID=:3ceh BOARD=quark_se_c1000_ss_devboard ARCH=arc  O=outdir-w28a-3ceh-quark_se_c1000_ss_devboard initconfig'
build output: Using /zephyr/zephyr/boards/arc/quark_se_c1000_ss_devboard/quark_se_c1000_ss_devboard_defconfig as base
build output: Merging /home/rrahulkx/zephyr/zephyr/tests/include/test.config
build output: Merging prj.conf
build output: Merging outdir-w28a-3ceh-quark_se_c1000_ss_devboard/arc_target.conf
build output: Merging outdir-w28a-3ceh-quark_se_c1000_ss_devboard/boot_stub_config.conf
build output: #
build output: # configuration written to .config
build output: #
build passed: 'make -j -C zephyr/zephyr/tests/booting/stub 
EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:3ceh BOARD=quark_se_c1000_ss_devboard 
ARCH=arc  
O=outdir-w28a-3ceh-quark_se_c1000_ss_devboard'

build passed

deploy passed

Reset

**console output :--**

eval failed: qc1000-01:default` at 3.45s
console output: starting test - test init
console output: FAIL - test_init@286. Cannot add multicast IPv6 address ff02::1
console output: failed
console output: starting test - IPv6 compare prefix
console output: passed
console output: starting test - IPv6 neighbor lookup fail
console output: passed
console output: starting test - IPv6 add neighbor
console output: passed
console output: starting test - IPv6 neighbor lookup ok
console output: passed
console output: starting test - IPv6 send NS extra options
console output: passed
console output: starting test - IPv6 send NS no options
console output: passed
console output: starting test - IPv6 handle RA message
console output: passed
console output: starting test - IPv6 parse Hop-By-Hop Option
console output: passed
console output: starting test - IPv6 change ll address
console output: passed
console output: starting test - IPv6 prefix timeout
console output: passed

console output: PROJECT EXECUTION FAILED

jukkar, 2017-10-19:
This worked just fine for me when running manually.

Could you try to increase the multicast address count to bigger value in prj.conf file
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=5
and report if that solves the issue?

gurucharanx, 2017-11-24:
@jukkar  this is running fine on qemu_x86 and arduino_101:x86 after adding the  CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=5 in the prj.conf file. 
on the commit ID: cb9390c7

Also it works fine on frdm_k64f :arm and sam_e70_xplained:arm boards after adding 
arch_whitelist: x86 arm in testcame.yaml file.

So I request to modify the corresponding files accrodingly

PR #5902 (2018-02-01): tests: net: ipv6: Increase the multicast address count
Increasing the IPv6 multicast address count from 2 to 5 so that
tests will always run ok.

Fixes #4401
commit 17fcbb05 (2018-02-01): tests: net: ipv6: Increase the multicast address count

Increasing the IPv6 multicast address count from 2 to 5 so that
tests will always run ok.

Fixes #4401
closed 2018-02-01 by jukkar