zephyrproject-rtos/zephyr #4412: Replace STM32 USB driver with DWC

galak, 2017-10-18:
The STM32 devices use a designware USB block so we should utilize the dwc driver for the device.

erwango, 2017-10-19:
To be exact: "The STM32 devices use designware USB block**s**"
There are actually variety of USB IPs across STM32 families. All of them being DW but different versions.
On the whole family, there are 4 "lines" of USB IPs. In each line having different versions.
As example of "line": USB DEV FS (F1 family), USB OTGS FS, USB OTG HS and one more which is USB with clock recovery (don't know how much one is different to the others). All of these had independant evolution in time, hence different versions.

Cube HAL API is made to handle this "heterogeneity". And it is possible to provide a single HAL based driver for the while family. From the starting point of having it working on one line, it will take minor steps to add support on other lines.

Since this abstraction is not available in DWC driver, I assume this would require more substantial efforts to get it supporting all these lines.

Hence my proposal is to keep Cube based driver to enable quick deployment on whole STM32 families. In a parallel track there could be some work to get dwc driver work on STM32, and user should chose which driver he wants to use.

carlescufi, 2019-10-09:
@erwango agreed that Cube seems like the better option here.
@galak any additional input, or should we close this?

erwango, 2020-05-25:
I'm closing that point as there was no further request nor interest on that point.
Please reopen if need be

closed 2020-05-25 by erwango