zephyrproject-rtos/zephyr #4416: add backtrace support when we have a crash

galak, 2017-10-18:


SebastianBoe, 2017-10-19:
Hi, just my two cents.

Unwinding the stack is tricky.

A halfway solution is to at least run addr2line to get the source code line
that triggered the crash. This will often be good enough.

I saw libunwind was suggested as a way to implement this, I'm not familiar with it,
but one way to do this with the tools we already depend on is to simply connect GDB
and dump the backtrace after the crash.

But both of these techniques would require that we stop running qemu / flasher scripts
directly, but instead use some kind of wrapper script like what marti did with the flashers
 (Something we probably want to do longterm anyway).

andrewboie, 2017-10-19:
the symbol table is not currently available at runtime, so having comprehensible output without getting GDB into the picture will require solving this in some way.

for most HW platforms, we convert the .elf into a .bin or .hex which has all debug stuff stripped out.

QEMU does get passed a .elf binary to run the kernel, but I haven't seen any easy way for the running kernel to access the elf's debug sections.

what we probably need is some non-default option to include some subset of the symbol table somewhere in the kernel's read-only data section. and then some kind of stack unwinding code to print it out. we don't care about footprint for QEMU targets and could enable this option by default for them.

i know the Linux kernel can do this with dump_stack(), I don't know details, but their approach is probably quite worth looking at.

nashif, 2017-12-08:
@galak can you provide a more detailed description for this request, this will narrow it down at least.

andrewboie, 2018-01-29:
https://github.com/DataAnalysisCosby/blog/blob/master/org/unwind.org

andrewboie, 2018-01-29:
https://github.com/CurieBSP/main/blob/master/bsp/src/machine/soc/intel/quark_se/quark/panic.c

andrewboie, 2018-01-29:
@lpereira  and I were just talking about this.

lpereira, 2018-03-16:
FWIW, this has been implemented for x86.

dleach02, 2018-05-02:
I'll also add that there is additional debug capabilities that I tend to add in to enable Micro Trace Buffer support in Arm cores that support it. Something to consider because it is useful when you take a fault to be able to look back at the instruction flow that got you where you are...

zhangshengping, 2019-03-12:
Is there some book introduce elf debug symbols? I am interested in this topic！

andrewboie, 2019-03-12:
unwinding the stack on x86 to get function addresses at runtime was pretty easy.

unwinding on ARM is much more involved: https://wiki.linaro.org/KenWerner/Sandbox/libunwind?action=AttachFile&do=get&target=libunwind-LDS.pdf

palavrov, 2023-02-10:
Hello,

I'll work on this.

Technically it's not complicated when the C compiler generates chained function stack frames  with -fno-omit-frame-pointer

Unfortunately for Thumb2, gcc doesn't generate predictable frame pointer chaining because function stack frame structure could be taken from dwarf debug tables like what libunwind does. That works for linux but doesn't work for embedded devices who strip the elf from all debug meta data and put it in flash. The only option is to use clang or to switch to arm32 where the function stack frame chaining is predictable.

There are lot of discussions about this like this one for example: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=92172

The output of the call stack will be prepared for addr2line i.e. something like `addr2line -safe filename.elf 0x01234567 0x89ABCDEF ...`

Theoretically it is possible to use `-mpoke-function-name` and then to look and print function names but from my experience this increases the binaries too much and doesn't worth the efforts.

Another annoying problem is that in the function stack frame we have the address where to return not the address of the caller i.e. addr2line will point to wrong source lines (especially with `-O3`) but still it will help to understand the reason for exception or bug. Unfortunately Thumb2 could use both 16-bit and 32-bit call instructions so it is far from trivial to properly calculate the address of the caller.

The same approach could be used for other architectures but I cannot test without access to dev boards so will keep the implementation only for ones I can test.

nashif, 2023-12-19:
@palavrov  

> I'll work on this.

any progress there?

palavrov, 2023-12-20:
Hey, indeed I left this task too deeply in my mental todo list i.e. almost forgot about it.

Right now I'm too busy with my customers projects and can't predict when will have enough time to work on this unless it becomes part of one them (which is s highly likely and it's almost an year like that and still didn't start working on this one).

Let me try to find some time next couple of weeks. If someone would like to work on this I could help with reviewing, discussing, testing ...

KozhinovAlexander, 2025-05-03:
Does anyone considered https://github.com/libunwind/libunwind?

palavrov, 2025-05-05:
Libunwind as far I know relies on stack frame information obtained from the .elf file at runtime i.e. no way to have it in stripped microcontroller flash binary. I just end my current project and while looking for new one will try to implement and close this one these days. @KozhinovAlexander if you need help to debug something we could collaborate as working on real case always help i.e. I need your feedback to craft better implementation that helps in real world cases.

KozhinovAlexander, 2025-05-05:
> Libunwind as far I know relies on stack frame information obtained from the .elf file at runtime i.e. no way to have it in stripped microcontroller flash binary. I just end my current project and [...]

Thank you a lot for your feedback. I am just stepping into libunwind at the moment. My first try was to ask ChatGPT if there is a Free RTOS port of libunwind. Thus needed to double-check it here :)

mingl0280, 2025-07-09:
I googled a little bit about stack unwinding progress in Zephyr, and it led me here. So I'm just sharing what I did to unwind the call stack during the execution...

Anyway, AFAIK, what's possible on some STM (ARM Cortex M4) CPU is: 

- Find the code section in Flash and acquire the main stack address 
- Get the current SP register value
- Scan the main stack from the SP address up to the end of the main stack:
- Try to see if the location is a link register location by doing this: 
```C
		/* the *sp value may be LR, so need to decrease a word to PC */
		pc = (*int_to_ptr(sp)) - sizeof(uint32_t);
		/* the Cortex-M using thumb instruction, so the pc must be an odd number */
		if (pc % 2 == 0) {
			continue;
		}
		/* fix the PC address in thumb mode */
		pc = *(int_to_ptr(sp)) - 1;
		if ((pc >= code_start_addr + sizeof(size_t)) && (pc <= code_start_addr + code_size)
		    && (depth < MAX_STACK_DEPTH)
		    /* check the instruction before PC address is 'BL' or 'BLX' */
		    && disassembly_ins_is_bl_blx(pc - sizeof(size_t)) && (depth < size)) {
			/* The second depth function may already be saved, so we need to ignore repeats */
			if ((depth == 2) && regs_saved_lr_is_valid && (pc == buffer[1])) {
				continue;
			}
			buffer[depth++] = pc;
		}
```
where `disassembly_ins_is_bl_blx` is basically checking if the given address is a valid BL or BLX address (here it's the simulated PC address - 4). 

This code only works when the code is on the main stack. This only provides the code address (where you can convert the addresses using addr2line afterwards).