zephyrproject-rtos/zephyr #4419: 6LoWPAN - source address uncompress corner case

pedropaulomartucci, 2017-10-18:
I'm new to zephyr, and trying to setup a server/client qemu environment to validate 802.15.4 and 6LoWPAN stacks.

For this purpose I'm using samples/ieee802154 with **CONFIG_NET_6LO_CONTEXT** disabled.

I started debugging and reached the state where the client send a packet with undefined source address:

```
[net/ipv6] [DBG] net_ipv6_send_ns: (0x200000e4): Sent Neighbor Solicitation from :: to ff02::1:ff9b:118d, target fe80::210:2030:9b:118d
[net/ieee802154] [DBG] ieee802154_compute_header_size: (0x200000e4): Broadcast destination
[net/ieee802154] [DBG] ieee802154_compute_header_size: (0x200000e4): Computed size of 15
[net/6lo] [DBG] compress_tfl: (0x200000e4): Trafic class and Flow label elided
[net/6lo] [DBG] compress_sa: (0x200000e4): SAM_00, SAC_1 unspecified src address
[net/6lo] [DBG] compress_da_mcast: (0x200000e4): M_1 dst is mcast
[net/6lo] [DBG] compress_da_mcast: (0x200000e4): DAM_01 6 bytes: 2nd byte + last five bytes
[net/6lo] [DBG] compress_IPHC_header: (0x200000e4): next header is not UDP (58)
```

This way, SAC=1 and SAM=00. Exactly like [RFC6282](https://tools.ietf.org/html/rfc6282#section-3.1.1) defines for undefined addressess (::).

When the server receives this packet, it starts to decompress:

```
[net/ieee802154] [DBG] validate_payload_and_mfr: (0x200000e4): Header size: 15, vs total length 48: payload size 33
[net/ieee802154] [DBG] ieee802154_reassemble: (0x200000e4): No frag dispatch (7b)
[net/6lo] [DBG] uncompress_tfl: (0x200000e4): Tcl and Flow label elided
[net/6lo] [DBG] uncompress_sa: (0x200000e4): SAC_0
[net/6lo] [DBG] uncompress_sa: (0x200000e4): SAM_00 full src addr inlined
```

There we can see that **uncompress_sa** handled SAC bit as 0, even when i checked `pkt->frags->data` with gdb:

```
(gdb) x/33xb pkt->frags->data
0x4057df <_net_buf_rx_bufs+31>:	0x7b	0x49	0x3a	0x02	0x01	0xff	0xde	0x08
0x4057e7 <_net_buf_rx_bufs+39>:	0xa9	0x87	0x00	0x18	0xd9	0x00	0x00	0x00
0x4057ef <_net_buf_rx_bufs+47>:	0x00	0xfe	0x80	0x00	0x00	0x00	0x00	0x00
0x4057f7 <_net_buf_rx_bufs+55>:	0x00	0x02	0x10	0x20	0x30	0x2e	0xde	0x08
0x4057ff <_net_buf_rx_bufs+63>:	0xa9
```
Converting `0x49` to binary we should have `01001001`, parsing it to IPHC fields:
- CID: 0
- SAC: 1
- SAM: 00
- M: 1
- DAC: 0
- DAM: 01

Taking a look in **uncompress_sa**, I noted that it doesn't check SAC to decompress the source address:

```
static inline u8_t uncompress_sa(struct net_pkt *pkt,
				 struct net_ipv6_hdr *ipv6,
				 u8_t offset)
{

	NET_DBG("SAC_0");

	switch (CIPHC[1] & NET_6LO_IPHC_SAM_11) {
	case NET_6LO_IPHC_SAM_00:
		NET_DBG("SAM_00 full src addr inlined");
		...
	case NET_6LO_IPHC_SAM_01:
		NET_DBG("SAM_01 last 64 bits are inlined");
		...
	case NET_6LO_IPHC_SAM_10:
		NET_DBG("SAM_10 src addr 16 bit compressed");
		...
	case NET_6LO_IPHC_SAM_11:
		NET_DBG("SAM_11 generate src addr from ll");
		...
	}

	return offset;
}
```

This way the source address is wrongly decompressed because **uncompress_sa** always handle SAM bits considering SAC=0. 

In some cases I got a fault segment error, because of negative value for `pkt->frags->len - offset` during the `memmove(pkt->frags->data, pkt->frags->data + offset, pkt->frags->len - offset)`.

I tried a workaround, just to make the execution stop failing, I don't know if it's a right solution:

```
static inline u8_t uncompress_sa(struct net_pkt *pkt,
				 struct net_ipv6_hdr *ipv6,
				 u8_t offset)
{
	if ((CIPHC[1] & NET_6LO_IPHC_SAC_1) == NET_6LO_IPHC_SAC_1 && 
            (CIPHC[1] & NET_6LO_IPHC_SAM_11) == NET_6LO_IPHC_SAM_00 ) {
		NET_DBG("SAC_1");
		NET_DBG("SAM_00 src addr undefined");
		return offset;
	}

	NET_DBG("SAC_0");

	switch (CIPHC[1] & NET_6LO_IPHC_SAM_11) {
	case NET_6LO_IPHC_SAM_00:
		NET_DBG("SAM_00 full src addr inlined");
		...
	case NET_6LO_IPHC_SAM_01:
		NET_DBG("SAM_01 last 64 bits are inlined");
		...
	case NET_6LO_IPHC_SAM_10:
		NET_DBG("SAM_10 src addr 16 bit compressed");
		...
	case NET_6LO_IPHC_SAM_11:
		NET_DBG("SAM_11 generate src addr from ll");
		...
	}

	return offset;
}
```
It apparently worked, and I could use shell net to ping between server and client.

Am I missing something?

pfalcon, 2017-10-19:
This appears to be addressed by https://github.com/zephyrproject-rtos/zephyr/pull/4432

jukkar, 2017-10-19:
By random luck this was encountered yesterday while working with other bug, the #4432 should fix this.

pedropaulomartucci, 2017-10-19:
Sounds good to me. Thank you.

pedropaulomartucci, 2017-10-20:
Closing the issue. Fixed by  #4432.

closed 2017-10-20 by pedropaulomartucci