zephyrproject-rtos/zephyr #4421: net: Duplicated functionality between net_pkt_get_src_addr() and net_context.c:create_sockaddr()

pfalcon, 2017-10-18:
In my review of https://github.com/zephyrproject-rtos/zephyr/pull/1055#issuecomment-331155034 , I expressed my surprised that we didn't have a func like net_pkt_get_src_addr() before. And indeed, we had something like that as net_context.c:create_sockaddr(), except that latter was confusingly named and worked only for TCP (kinda, the only thing it needs from TCP are port numbers, and those match UDP header).

Let's now think how to get rid of duplication.

pfalcon, 2017-10-18:
Note that net_pkt_get_src_addr() almost certainly incorrectly uses net_pkt_tcp_data() to access TCP/UDP headers. Almost, unless @askawu have thought of that and a net data fragment guaranteed to be even (which probably can't be, e.g. `default 125 if NET_L2_IEEE802154` (it's strange it's not 127 btw)).

pfalcon, 2017-10-18:
See https://github.com/zephyrproject-rtos/zephyr/pull/4422

pfalcon, 2017-10-18:
My proposal would be to base on net_pkt_get_src_addr(), fix it, and drop sockaddr_len param - it and associated checks are superfluous and only add overhead.

pfalcon, 2017-10-20:
@askawu : Please let me know if you agree with the changes proposed above and if you're interested to implement them. Otherwise, I'll go for them.

askawu, 2017-10-24:
@pfalcon, net_pkt_get_src_addr() has been fixed by #4484. I will work on replacing create_sockaddr().

commit 1bdd80f8 (2017-11-06): net: context: Replace net pkt parsing by helper functions

The code parsing received net pkt to get source or destination
sockaddr repeats multiple times in net_context.c.

Eliminate the duplication by net_pkt_get_src_addr() and
net_pkt_get_dst_addr() which can handle different internet protocol
(i.e. ipv4 or ipv6) and transport protocol (i.e. tcp or udp)

Fixes: #4421
closed 2017-11-06 by jukkar