zephyrproject-rtos/zephyr #4424: Turning on network debug message w/ LwM2M sample client will result in stack check failure

rbtchc, 2017-10-19:
Test against master branch
commit ee107baad9c438d24c6fd584f27c7d133acbbedc
Author: Johan Hedberg <johan.hedberg@intel.com>
Date:   Wed Oct 18 10:29:38 2017 +0300

Enable network debug info by adding **CONFIG_NET_LOG_GLOBAL=y** to file prj_qemu_x86.conf
<pre>
diff --git a/samples/net/lwm2m_client/prj_qemu_x86.conf b/samples/net/lwm2m_client/prj_qemu_x86.conf
index 84fa4da..c2b82da 100644
--- a/samples/net/lwm2m_client/prj_qemu_x86.conf
+++ b/samples/net/lwm2m_client/prj_qemu_x86.conf
@@ -41,4 +41,6 @@ CONFIG_NET_APP_PEER_IPV6_ADDR="2001:db8::2"
 CONFIG_NET_APP_MY_IPV4_ADDR="192.0.2.1"
 CONFIG_NET_APP_PEER_IPV4_ADDR="192.0.2.2"
 
+CONFIG_NET_LOG_GLOBAL=y
+
 CONFIG_NET_SLIP_TAP=y
</pre>

The error is met shortly after I execute command "make run"
<pre>
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3303/0/5700, value:0x0042f3bc, len:8
[net/ctx] [DBG] net_context_bind: (0x00407f00): Context 0x004032a0 binding to UDP [2001:db8::1]:63532 iface 0x00436c40
[net/app] [DBG] _net_app_print_info: (0x00407f00): net app connect [2001:db8::1]:63532 -> [2001:db8::2]:5683
[net/conn] [DBG] net_conn_register: (0x00407f00): [0/10/17/0x33] remote 0x004032b4/2001:db8::2/5683 local 0x0042f2d8/2001:db8::1/63532 cb 0x00007910 ud 0x00400040
[lib/lwm2m_rd_client] [INF] lwm2m_rd_client_start: LWM2M Client: qemu_x86
[lib/lwm2m_rd_client] [INF] sm_do_init: RD Client started with endpoint 'qemu_x86' and client lifetime 0
[net/ipv6] [DBG] net_ipv6_prepare_for_send: (0x004080a0): No default route to 2001:db8::2
[net/ipv6] [DBG] net_ipv6_prepare_for_send: (0x004080a0): Neighbor lookup 0x00000000 (255) iface 0x00436c40 addr 2001:db8::2 state -
[net/nbr] [DBG] net_nbr_ref_debug: (0x004080a0): nbr 0x00436080 ref ***** Stack Check Fail! *****
Current thread ID = 0x004080a0
Faulting segment:address = 0x0008:0x00002caf
eax: 0x00436a48, ebx: 0x00000020, ecx: 0x00023740, edx: 0x00000031
esi: 0x00426094, edi: 0x00003f01, ebp: 0x00426004, esp: 0x00426000
eflags: 0x216
Fatal fault in thread 0x004080a0! Aborting.
QEMU: Terminated
</pre>

Explicitly setting **CONFIG_USERSPACE=n** in prj_qemu_x86.conf and the error will be gone.

nashif, 2017-10-19:
I guess this is caused by HW_STACK_PROTECTION and not a userspace issue, @andrewboie fyi

rbtchc, 2017-10-19:
@nashif you are right. it's related to HW_STACK_PROTECTION instead USERSPACE.

rbtchc, 2017-10-19:
It turns out I only have to expand the stack size of CONFIG_LWM2M_RD_CLIENT_STACK_SIZE when enabling the debug info.

PR #4428 (2017-10-19): net: lwm2m: increase rd client stack size when NET_LOG_GLOBAL=y
The stack of rd client is exhausted while running lwm2m client w/ IPv6
and network log global enabled. Increase the stack size to 1536 when
NET_LOG_GLOBAL is enabled.

Fixes #4424
closed 2017-10-19 by rbtchc