zephyrproject-rtos/zephyr #4429: I2C: stm32-i2c-v2 Driver (F0/F3/F7) gets stuck in endless loop when handling restart conditions

DWagenk, 2017-10-19:
in drivers/i2c/i2c_ll_stm32_v2.c:

When an i2c-message prepends a restart condition (I2C_MSG_RESTART flag is set) in function **msg_init** auto end mode is beeing disabled for this message, but function **msg_done** gets stuck in a loop forever, waiting for a stop condition on the bus.

Tested with stm32f3_disco board and samples/drivers/i2c_fujitsu_fram modified to use I2C_1

ydamigos, 2017-10-19:
@DWagenk Thanks for reporting. I will check it. I have done i2c_burst_reads (which sets I2C_MSG_RESTART flag) multiple times without facing such an issue.

DWagenk, 2017-10-19:
@ydamigos Just checked it with burst_read function, the problem stays the same.
I also just noticed, that stm32-i2c-(v1 and v2) internally use the I2C_MSG_RESTART flag different from the API description.
API: I2C_MSG_RESTART = This message starts with a restart condition
STM32: I2C_MSG_RESTART = This message is followed by a restart condition

ydamigos, 2017-10-19:
@DWagenk Does it loop forever in write_bytes or read_bytes function (as they are named in main.c of i2c_fujitsu_fram sample)?

DWagenk, 2017-10-19:
Sorry, my first post was not completely acurate, I've been looking at the same problem for too long.

I get stuck during write operation. It sends Message 0 (main.c line 34-36), then in msg_done it waits for the stop condition (i2c_ll_stm32_v2.c line 60) and afterwards issues another stop condition (Line 63). 
When trying to send Message 1 [main.c line  39-41) it gets stuck in stm32_i2c_msg_write waiting for TXIS or NACK flag (i2c_ll_stm32_v2.c line 206). When I look into the registers with the debugger theres no flag except STOPF set. 

So I guess theres some time correlation and the 2nd stop condition prevents TXIS or NACK events from happening.

This part gets fixed by following patch

--- a/drivers/i2c/i2c_ll_stm32_v2.c
+++ b/drivers/i2c/i2c_ll_stm32_v2.c
@@ -60,7 +60,7 @@ static inline void msg_done(struct device *dev, unsigned int flags)
 		while (!LL_I2C_IsActiveFlag_STOP(i2c)) {
 			;
 		}
-		LL_I2C_GenerateStopCondition(i2c);
+		LL_I2C_ClearFlag_STOP(i2c);
 	}

But then theres other errors in read transactions. I'll post more on that.

DWagenk, 2017-10-19:
With the patch from my previous post I've got the following problems with read transactions:

I expect: 
(with <S> as start or repeated start, <P> as stop condition)
<S><SLAVE_ADDR|W><FRAM_ADDR_HIGH><FRAM_ADDR_LOW><S><SLAVE_ADDR|R><BYTE_0>[...]
<BYTE_n><P>
I get:
<S><SLAVE_ADDR|W><FRAM_ADDR_HIGH><FRAM_ADDR_LOW><P><S><SLAVE_ADDR|R><BYTE_0>[...]<BYTE_n>______timeout/next transfer

Diagnose:
During Message 0 (FRAM ADDR) AutoEndMode is disabled. Now while preparing Message 1 (read bytes) AutoEndMode gets enabled again. This happens to fast -> transmission of last byte from Message 0 is not finished yet. When message 0 finishes then AutoEndMode is already turned on again, so it issues a stop condition. The read transfer afterwards misses the stop condition, because it needs to be triggered manually via the LL_I2C_GenerateStopCondition function.

Best guess for a solution: We need to wait for TC Flag beeing set before sending Message 1.

DWagenk, 2017-10-19:
Another hint: the missing stop conditions after read transactions disappear when I set a breakpoint before LL_I2C_Disable(i2c); in i2c_ll_stm32.c line 102.

So theres one problem with the unwanted 
P - S 
where we want a single repeated start
S 
between write and read message. and another problem with stop conditions becoming lost, because I2C is beeing disabled without waiting for the stop condition.

ydamigos, 2017-10-20:
> I expect:
> (with S as start or repeated start, P as stop condition)
> S - SLAVE_ADDR|W - FRAM_ADDR_HIGH - FRAM_ADDR_LOW - S - SLAVE_ADDR|R - BYTE_0 - [...] -
> BYTE_n - P
> I get:
> S - SLAVE_ADDR|W - FRAM_ADDR_HIGH - FRAM_ADDR_LOW - P - S - SLAVE_ADDR|R - BYTE_0 - [...] -
> BYTE_n timeout/next transfer

I checked the manual of the FRAM (Fujitsu MB85RC256V). The Write Page command should be:
S - SLAVE_ADDR|W - FRAM_ADDR_HIGH - FRAM_ADDR_LOW - BYTE_0 - [...]  BYTE_n - P
There is no STOP or REPEATED start condition between the FRAM address and the data.

It seems that the I2C API expects a different behavior from the driver than handling a Message as:
S - SLAVE_ADDR|W - DATA - P or Sr

It is like the data in Message 1 in i2c_burst_write are appended as data if there is no I2C_MSG_RESTART (Message 1) or I2C_MSG_STOP (Message 0) flags:
Message 0: S - SLAVE_ADDR|W - FRAM_ADDR_HIGH - FRAM_ADDR_LOW
Message 1: BYTE_0 - [...]  BYTE_n - P

I don't know if this is the expected behavior from the I2C API. I will have to check the I2C driver implementation of the vendors.

DWagenk, 2017-10-20:
judging from the API doc

> I2C_MSG_STOP (1 << 1)
Send STOP after this message.
I2C_MSG_RESTART (1 << 2)
RESTART I2C transaction for this message.

I'd say this is expected behaviour for the I2C API. If first message doesn't have I2C_MSG_STOP set and second message doesn't have I2C_MSG_RESTART set they appear as one long message on the bus. I actually need this functionality, otherwise I have to copy data back and forth to connect a multi-part-message into one big message.

I'll check with a frdm_kw41z board later how it behaves using the i2c_burst_write functions.

DWagenk, 2017-10-20:
@ydamigos 
I've got my original problem (endless loop) solved and tested. I'll submit the fix as a pullrequest and close this issue. 
Looking into the source of stm32-i2c-v1 driver the problem with API and driver not beeing consistend exists there too, so I'd move that to a seperate issue.
OK?

DWagenk, 2017-10-20:
OK, reopend

PR #4458 (2017-10-23): i2c: fix issue of stm32-i2c-v2 (F0/F3/F7) driver hanging in endless loop
Fixes #4429
Driver didn't work properly when a transfer consisted of multiple
messages.
Tested with stm32f3_disco board and samples/drivers/i2c_fujitsu_fram
example adapted to use I2C_1 as I2C_DEV
commit 30bed0ff (2017-10-23): i2c: fix issue of stm32-i2c-v2 (F0/F3/F7) driver hanging in endless loop

Fixes #4429
Driver didn't work properly when a transfer consisted of multiple
messages.
Fix doesn't use auto end mode anymore. msg_done function waits for
transfer to complete and issues stop condition if necessary.

Tested with stm32f3_disco board and samples/drivers/i2c_fujitsu_fram
example adapted to use I2C_1 as I2C_DEV
PR #4508 (2017-10-24): i2c: fix issue of stm32-i2c-v2 (F0/F3/F7) driver hanging in endless loop
Fixes #4429
Driver didn't work properly when a transfer consisted of multiple
messages.
Fix doesn't use auto end mode anymore. msg_done function waits for
transfer to complete and issues stop condition if necessary.

Tested with stm32f3_disco board and samples/drivers/i2c_fujitsu_fram
example adapted to use I2C_1 as I2C_DEV
commit 695acc8e (2017-10-24): i2c: fix issue of stm32-i2c-v2 (F0/F3/F7) driver hanging in endless loop

Fixes #4429
Driver didn't work properly when a transfer consisted of multiple
messages.
Fix doesn't use auto end mode anymore. msg_done function waits for
transfer to complete and issues stop condition if necessary.

Tested with stm32f3_disco board and samples/drivers/i2c_fujitsu_fram
example adapted to use I2C_1 as I2C_DEV
closed 2017-10-23 by DWagenk