zephyrproject-rtos/zephyr #4441: shell: Add option for wildcard command

KaSroka, 2017-10-19:
It will be nice to have some wildcard command in shell. This will allow to expose CLI of external modules without implementing every command in Zephyr.

After selecting desired module every typed command will be forwarded to wildcard command handler except the exit command.

nashif, 2018-01-15:
Can you please give more details? shell modules and commands do not need to be part of Zephyr, they can be implemented on the application level as well. See the sample shell application which does exactly that.

KaSroka, 2018-01-17:
I can give you an example from #4997.

OpenThread has multiple commands e.g. status, channel, ipaddr (it's really small subset shown here).
With current approach I can implement it in two ways:
1. Implement every OT command as Zephyr shell command - lots of unnecesary work and also we would have to keep track of OT commands and align Zephyr clones.
2. Implement one dummy command which arg is real command for OpenThread. This approach is currently implemented. It has the disadventage that dummy command name has to be prepended to every OT command. So lets say modlue name is 'ot' and dummy command is 'cmd' we have to type `ot cmd state` or after selecting ot module `cmd state`. 

I would like to add possibility of wildcard command - tell Zephyr shell that this module has handler for e.g. any unknown command in this module. This will allow to ommit `cmd` part from previous examples, as I could catch all commands for ot module in this handler and forward it to OpenThread CLI module. So we could simply type `ot state` or after selecting 'ot' module just `state`.

rlubos, 2019-01-22:
Closing this one, as we've achieved requested functionality with new shell module and #12345.

closed 2019-01-22 by rlubos