zephyrproject-rtos/zephyr #4445: sanitycheck --platform-limit is broken

andrewboie, 2017-10-19:
Before the YAML transition, there used to be per-arch .ini files which indicated a priority ordering for platforms within a certain arch.

This ordered list was used to choose what platform to use with a limited set of platforms to build.

This is now broken. There is no ordered list of platforms any more, instead there is a "default" tag added to one platform. This is not a good change of design.

- If the "default" platform can't run a test, how is an alternative chosen? The old way was to filter the ordered list of platforms for those that the test can actually run on, and then choose the first N platforms of that set, where N was the parameter to --platform-limit
- sanitycheck -L2 or -L3 don't choose the top 3 platforms per test to run it on, seems to choose almost the same set of tests now, it's broken

nashif, 2017-10-19:
`--platform-limit` had a design issue that did not allow us to select multiple boards per architecture for example when the value was set 1, this a problem for example on ARM where we wanted to test and build for both cortex-m0 and cortex-m3/4. Now we can define multiple boards per architecture as default.

> If the "default" platform can't run a test, how is an alternative chosen? The old way was to filter the ordered list of platforms for those that the test can actually run on, and then choose the first N platforms of that set, where N was the parameter to --platform-limit

That feature is indeed broken and useful to have, will look into that.

> sanitycheck -L2 or -L3 don't choose the top 3 platforms per test to run it on, seems to choose almost the same set of tests now, it's broken

I confirm this is broken,  is there a need for this however? The main reason behind this feature was to allow tuning how many tests we run in CI which was taking too long in Jenkins. Now we can also have multiple defaults per architecture, so some use cases of this are covered when defining default.

andrewboie, 2017-10-19:
> I confirm this is broken, is there a need for this however? The main reason behind this feature was to allow tuning how many tests we run in CI which was taking too long in Jenkins. Now we can also have multiple defaults per architecture, so some use cases of this are covered when defining default.

If we don't think we need this any more, I would consider removing --platform-limit option to be adequate.

commit ab74706e (2017-12-11): sanitycheck: remove platform-level option

Since the move to YAML format and the change in how we define default
platforms this is no longer needed as we are able to set multiple
default platforms per architecture and not using a list based on
priority anymore.

Fixes #4445
closed 2017-12-11 by nashif