zephyrproject-rtos/zephyr #4476: Multiple build failures with i2c_ll_stm32.c driver

nashif, 2017-10-21:
see https://app.shippable.com/github/zephyrproject-rtos/ci-test/runs/923/7/tests

```
  CC      kernel/thread.o
  CC      kernel/thread_abort.o
/home/buildslave/src/github.com/zephyrproject-rtos/ci-test/zephyr/drivers/i2c/i2c_ll_stm32.c:102:12: error: ‘i2c_stm32_init’ defined but not used [-Werror=unused-function]
 static int i2c_stm32_init(struct device *dev)
            ^~~~~~~~~~~~~~
/home/buildslave/src/github.com/zephyrproject-rtos/ci-test/zephyr/drivers/i2c/i2c_ll_stm32.c:97:36: error: ‘api_funcs’ defined but not used [-Werror=unused-const-variable=]
 static const struct i2c_driver_api api_funcs = {
                                    ^~~~~~~~~
cc1: all warnings being treated as errors
/home/buildslave/src/github.com/zephyrproject-rtos/ci-test/zephyr/scripts/Makefile.build:183: recipe for target 'drivers/i2c/i2c_ll_stm32.o' failed
make[5]: *** [drivers/i2c/i2c_ll_stm32.o] Error 1
make[5]: Target '__build' not remade because of errors.
/home/buildslave/src/github.com/zephyrproject-rtos/ci-test/zephyr/scripts/Makefile.build:357: recipe for target 'drivers/i2c' failed
make[4]: *** [drivers/i2c] Error 2
  CC      drivers/interrupt_controller/exti_stm32.o
  CC      kernel/timer.o

```

closed 2017-10-23 by galak

erwango, 2017-10-23:
@galak: I wonder how we missed this at PR review. Is there a second wave of testing with a different test set done on main branch (or would it be a peridocal non regression sequence)?

galak, 2017-10-23:
We do a nightly build of everything which is what caught this, but not on PRs.  Its only done on things we have committed.