zephyrproject-rtos/zephyr #4481: Build failure with CONFIG_NET_DEBUG_APP=y

vakulgarg, 2017-10-23:
Sample applications fail to build on defining CONFIG_NET_DEBUG_APP=y in project configuration.
Error log with zephyr/samples/net/lwm2m_client/ is given below.

-------------------------------------------------------------------------------------------------------------
  LINK    zephyr.lnk
subsys/built-in.o: In function `net_app_connect':
/home/b16394/network_protocols/networkprotocols/WP1/zephyr/subsys/net/lib/app/client.c:599: undefined reference to `_net_app_select_net_ctx'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/b16394/network_protocols/networkprotocols/WP1/zephyr/Makefile:876: zephyr_prebuilt.elf] Error 1
make[2]: Leaving directory '/home/b16394/network_protocols/networkprotocols/WP1/zephyr/samples/net/lwm2m_client/outdir/qemu_x86'
make[1]: *** [Makefile:178: sub-make] Error 2
make[1]: Leaving directory '/home/b16394/network_protocols/networkprotocols/WP1/zephyr'
make: *** [/home/b16394/network_protocols/networkprotocols/WP1/zephyr/Makefile.inc:82: all] Error 2

mike-scott, 2017-10-23:
Hello, may I have the command you are using to build the app, SDK version and host OS / version?

vakulgarg, 2017-10-23:
I am compiling lwm2m_client after setting CONFIG_NET_DEBUG_APP=y using command 'make' inside zephyr/samples/net/lwm2m_client. The zephyr code is from HEAD of master branch. 

The host machine is fedora 26.
The SDK version is 0.9.1.

mike-scott, 2017-10-24:
@jukkar yep, I get this same error.

PR #4515 (2017-10-24): net: app: Fix compilation error if debugging is enabled
If user has enabled CONFIG_NET_DEBUG_APP, then the debugging
version of _net_app_select_net_ctx() was not properly declared
and it caused compile error.

Fixes #4481
commit 5eb8936b (2017-10-24): net: app: Fix compilation error if debugging is enabled

If user has enabled CONFIG_NET_DEBUG_APP, then the debugging
version of _net_app_select_net_ctx() was not properly declared
and it caused compile error.

Fixes #4481
closed 2017-10-24 by andrewboie