zephyrproject-rtos/zephyr #4503: CONFIG_STACK_SENTINEL inconsistencies

dleach02, 2017-10-24:
Using CONFIG_STACK_SENTINEL for thread stack debugging is not consistently applied.

- _interrupt_stack is not initialized when SENTINEL is enabled
- _stack_unused_space_get() function doesn't account for sentinels and as a result gives incorrect values

dleach02, 2017-10-24:
I have a PR being prepped to address this

andrewboie, 2017-10-24:
CONFIG_STACK_SENTINEL is not designed to catch overflows of the interrupt stack, only thread stacks.
We also have MPU-based stack overflow mechanisms for ARM which are much more robust.

dleach02, 2017-10-24:
@andrewboie , why not? Doesn't seem to be a reason not to while there does seem to be a debug reason to have it, particularly if you want to debug stack problems which I recently did. 

The reason it is highlighted is that checking for unused stack space doesn't account for the SENTINEL but when you add that in it borks on the interrupt stack because the function checking doesn't know what stack is being checked. 

It seems like if one enables CONFIG_STACK_SENTINEL that it should apply to all stacks in the system and not just a select few. 

Also, I'll add that it is not all that clear what should be enabled to have all of the differing checking mechanisms in play. In my case, I was debugging on an M0 core and had a situation where a stack was not big enough (not the interrupt stack). I started enabling different debug options to get better visibility and realized that there is this unhook on where CONFIG_STACK_SENTINEL is used and how you can globally check all the stack sizes.

The code to add the SENTINEL for ARM M0 was all of three lines of conditionally included assembly so I don't think it is that much of an overhead.

andrewboie, 2017-10-24:
> apply to all stacks in the system and not just a select few.

'select few'? it's every stack in the system except the interrupt stack, which has special handling. to actually catch stack overflows in interrupt mode you'd need additional hooks in the interrupt handling code to check it on the way out of an interrupt. are you volunteering to implement this on all 6 arches?

dleach02, 2017-10-25:
No I'm not. In fact, I didn't even do that in the code I changed. 

What I did was to ensure that the SENTINEL stamp was added to the interrupt stack. This then allowed the stack usage function to correctly identify how much of the stack was used instead of barfing on the missing SENTINEL and assuming that all of it was used. Note that what I'm talking about is two changes. The first was to change the stack usage function so that it is aware of the SENTINEL if this was configured. The second was to ensure that all the stacks had this SENTINEL if configured.

Now if folks found it useful to also check for overflows then I would suggest a new ticket for that. My ticket wasn't intended for that.

andrewboie, 2017-11-14:
Fixed by #4930

closed 2017-11-14 by andrewboie