zephyrproject-rtos/zephyr #4505: sanitycheck should distinguish compile errors from link errors

galak, 2017-10-24:
We have several CI cases in which we get link errors for code size growth because we enable something like DEBUG, or build with a different toolchain.  It would be nice to distinguish these errors (build_error in sanity check) from actual compile errors.

andrewboie, 2017-11-01:
You'd have to have Make pass a different return value if the build fails due to link in order for this to be possible, as the parallel build is controlled by Make and not sanitycheck.

nashif, 2019-06-19:
I do not think this will work, you might also have link errors due to some missing symbols and other reasons, trying to cover all cases is just going to complicate sanitycheck without any value. This issue is not that common and should be dealt with using filters instead.

closed 2019-06-19 by nashif

marc-hb, 2019-06-19:
> You'd have to have Make pass a different return value if the build fails due to link in order for this to be possible

The Makefile generated by sanitycheck could have three targets: build, link, run. Then sanitycheck could call each one in sequence. Of course this depends on CMake's ability to generate Makefiles or ninja files with such separate targets in the first place. It's probably the case already but with unpredictable target names.

andrewboie, 2019-06-19:
I do not think this is worth the effort

marc-hb, 2019-06-19:
I would not embark on such an adventure myself either; the value seems low to me too. However there may be a solution that doesn't increase complexity and if yes then I would be happy to review it if someone else cares deeply enough for some reason(s).