zephyrproject-rtos/zephyr #4533: IPv6 and IEEE802.15.4 Initialization Startup problem

dleach02, 2017-10-25:
The startup of an IPv6 with 802.15.4 is out of sync which causes the first couple of packets to be sent with an invalid destination 802.15.4 PAN ID of 0. This was also confirmed with Wireshark.

I observed in GDB and debug logging that at startup the 15.4 driver function to set the address, set TX power, set channel, and start() are called. The IPv6 stack then starts sending frames before the 15.4 stack sets the PAN ID. 

The IPv6 stack appears to have only one synchronization point with 802.15.4 via: 

init_iface() -> net_if_init() -> init_iface() 

Where it calls api->init(iface). This goes directly to the init function in the 15.4 driver and not to anything in the 15.4 stack. It is unclear to me right now how the 15.4 stack even initializes but it seems like the IP stack needs to wait till everything below it is initialized and ready.

tbursztyka, 2017-10-27:
A PAN ID of 0 is totally fine.

What you see here is the IPv6 DAD process running before you set your pan id, and once done: DAD is not re-ran again.

I'll come with something.

dleach02, 2017-10-27:
Yes, I saw what was running. Just seemed strange that the IP stack was able to send anything before the underlying network was fully "up". I may be looking at this wrong but I consider it a link up vs down condition where the protocol stacks should hold off until they get a link up indication. It may be though that this is different for 15.4. 

In the end, it didn't prevent things from running but frankly I didn't know what the side affect would be for the DAD process if the packets being sent are all being dropped by the receivers...

tbursztyka, 2018-03-15:
This was addressed long ago, now 15.4 L2 is following down/up policy.

closed 2018-03-15 by tbursztyka