zephyrproject-rtos/zephyr #4544: [BLE Mesh] Error: Failed to advertise using Node ID

SharanIoT, 2017-10-25:
We are trying to evaluate Zephyr RTOS for our projects. Currently I'm playing with BLE Mesh sample application using the SiLabs android app. With some tweaks I was able to provision a device. However just after provisioning firmware fails with an error: **Failed to advertise using Node ID**. 

Is it a known issue? What's the solution for this?

Thanks & Regards,
Sharan

jhedberg, 2017-10-25:
This isn't actually a critical thing. What's happening is that the code tries to re-enable connectable advertising (for Node ID), but since there's only support for a single connection the controller returns an error for this. As soon as the current GATT connection is gone the code should succeed in doing the needed advertising. So in practice you can just ignore this message. I agree however that we can improve the situation: either the message should be downgraded to a warning, or we should do a bit more tracking of supported connection counts so that we don't try to do connectable advertising when it's foreseeable that the controller will reject it.

jhedberg, 2017-10-25:
Could you confirm my suspicions (that this isn't actually preventing any essential functionality), so that I can close this issue?

jhedberg, 2017-10-25:
Btw, if you want to support multiple GATT connections, you should increase CONFIG_BT_MAX_CONN to something more than 1 (the default value), in which case the error should also go away.

SharanIoT, 2017-10-25:
Hi Johan, Got it. Thanks for quick answer.

closed 2017-10-25 by SharanIoT