zephyrproject-rtos/zephyr #4563: [BLE Mesh]:

SharanIoT, 2017-10-26:
Hello,
I'm trying to test the mesh sample app using SiLabs app. Provisioning did succeed. Now, I'm not very clear how to verify the functionalities -generic on/off and level. There is no clear explanation about the handling the callbacks. I could see the traces are being printed in the callbacks. But I don't know how to extract the date "set" data and how to respond to the "get". 

Thanks & Regards,
Sharan

jhedberg, 2017-10-26:
I'm not sure if filing issues is the best way for this kind of questions, since there really isn't any obvious bug here. Using the mailing list or our IRC channels may be a better option. Regarding these methods, they're indeed unimplemented (beyond simple printk calls). You could e.g. take examples of how to parse the data and send responses by looking at how the health and configuration models are implemented. You can find the code for them in subsys/bluetooth/host/mesh/cfg.c and subsys/bluetooth/host/mesh/health.c

closed 2017-10-26 by jhedberg

nashif, 2017-10-26:
An issue does not have to be a bug, it can also be a question. And having it here is way better than IRC. (Can be found later)

SharanIoT, 2017-10-27:
Hi Nashif,

I'm new to the Zephyr world. But I like Zephyr, at the moment it's the only true IoT platform addressing wireless standard and security infrastructure. But problem for me is lack of developer forum to post our queries. I'm believe that saves significant time for the new people who are trying to work with Zephyr.

Regards,
Sharan