zephyrproject-rtos/zephyr #4565: net_context_recv always fails with timeout=K_FOREVER

vakulgarg, 2017-10-26:
I am using synchronous net_context_recv with UDPv6 socket with infinite timeout.

I create a UDP net_context, bind it and do a synchronous net_context_recv().
On receiving a datagram, my callback gets invoked. 
Once callback returns, the net_context_recv() returns error code -ETIMEDOUT.

Looking at net_context_recv, the reason seems obvious.
At the end of function net_context_recv(), with non-zero timeout condition, when k_sem_take() returns '0' i.e. the semaphore can be acquired due to k_sem_give() in packet_received(), the  logic returns -ETIMEOUT even for the case when UDP datagram was received.

jukkar, 2017-10-26:
Indeed, good catch. I will send a patch shortly.

jukkar, 2017-10-26:
@vakulgarg Could you test #4571 in your setup so that we are sure I did not miss some corner case?

PR #4571 (2017-10-30): net: context: Make sure we honor timeout when waiting data
The call to net_context_recv() with timeout returned -ETIMEDOUT
even when data was returned properly and there was no timeout.

Fixes #4565
commit 0e7c8edc (2017-10-30): net: context: Make sure we honor timeout when waiting data

The call to net_context_recv() with timeout returned -ETIMEDOUT
even when data was returned properly and there was no timeout.

Fixes #4565
closed 2017-10-30 by jukkar