zephyrproject-rtos/zephyr #4567: [BLE Mesh]: Multiple elements in a node

SharanIoT, 2017-10-26:
Hello,

I'm trying to add multiple elements to mesh sample application as pasted the snippet below. I've used the location descriptors from 1 to 4 for demo. The issue is that the SiLabs apps fails to read the DCD and quits. I might be missing something. 

static struct bt_mesh_elem elements[] = {
#if 0
	BT_MESH_ELEM(0, root_models, vnd_models),
#else
	BT_MESH_ELEM(1, root_models, vnd_models),
	BT_MESH_ELEM(2, root_models, vnd_models),
	BT_MESH_ELEM(3, root_models, vnd_models),
	BT_MESH_ELEM(4, root_models, vnd_models),
#endif
};

jhedberg, 2017-10-26:
The models that you give to BT_MESH_ELEM need to be unique instances, since they track unique runtime states. So you can only give a model array once to a BT_MESH_ELEM macro.

jhedberg, 2017-10-26:
Also note that as you create a more complex device composition, the composition data will likely no-longer fit in the max three transport layer segments that the current sample configurations support. So you might need to increase that. For outgoing segments, the number is directly dependent on the number of mesh advertising buffers, so you'll want to set CONFIG_BT_MESH_ADV_BUF_COUNT to some higher value than the default 6 (hint: the code uses 'count - 3' as the max number of transport layer segments).

closed 2017-10-26 by jhedberg

Laczen, 2017-10-26:
As an example of a node with multiple elements (2) you can use the following code. The configuration server model (cfg_srv) is only allowed in the first element.

static struct bt_mesh_model root_models[] = {
	BT_MESH_MODEL_CFG_SRV(&cfg_srv),
	BT_MESH_MODEL_HEALTH_SRV(&health_srv),
	BT_MESH_MODEL(BT_MESH_MODEL_ID_GEN_ONOFF_SRV, gen_onoff_srv_op,
		      &gen_onoff_srv_pub, NULL),
        BT_MESH_MODEL(BT_MESH_MODEL_ID_GEN_ONOFF_CLI, gen_onoff_cli_op,
		      &gen_onoff_cli_pub, NULL),
};

static struct bt_mesh_model root_models1[] = {
        BT_MESH_MODEL_HEALTH_SRV(&health_srv),
	BT_MESH_MODEL(BT_MESH_MODEL_ID_GEN_ONOFF_SRV, gen_onoff_srv_op,
		      &gen_onoff_srv_pub, NULL),
        BT_MESH_MODEL(BT_MESH_MODEL_ID_GEN_ONOFF_CLI, gen_onoff_cli_op,
		      &gen_onoff_cli_pub, NULL),
};

static struct bt_mesh_model vnd_models[] = {};

static struct bt_mesh_elem elements[] = {
	BT_MESH_ELEM(0, root_models, vnd_models),
        BT_MESH_ELEM(1, root_models1, vnd_models),
};

SharanIoT, 2017-10-27:
Hi Laczen,

I tried the same thing, it didn't work. Could be a bug on the SiLabs app, always fails to read "DCD". 
Did you test the multiple elements scenario?

Thanks,
Sharan

Laczen, 2017-10-27:
Hi SharanIoT,

I just tried and indeed it does not work with the SiLabs app. Maybe there is no support for multiple elements in the app. It does work with meshctl.

Kind regards,

Jehudi

SharanIoT, 2017-10-30:
Hi Jehudi,

Thats a useful info! I think I need to setup a linux machine for Mesh Testing :)

Thanks & Regards,
Sharan

shukla-ashish-92, 2017-11-25:
Hi Sharan, 
I've been trying to provision and configure /Bluetooth/mesh with Android app by silabs. After disabling the OOB feature, it successfully provision the node, but configuration page never appears as shown in the video. If you've made it work, would you please share hacks you did to configure it ?

SharanIoT, 2017-11-26:
Hi Ashish,

How did you disable the OOB feature? how did you initialize bt_mesh_prov and bt_mesh_cfg initialize?

Thanks & Regards,
Sharan