zephyrproject-rtos/zephyr #4568: dts generation incorrect for frdm_k64

galak, 2017-10-26:
We end up generating something like:

/* i2c@40066000 */
#define NXP_KINETIS_I2C_40066000_BASE_ADDRESS_0         0x40066000
#define NXP_KINETIS_I2C_40066000_BASE_ADDRESS_1         0x1000

This is wrong, this is probably due to miss handling of addr/size cells in the i2c bus controller node.

This should be:

/* i2c@40066000 */
#define NXP_KINETIS_I2C_40066000_BASE_ADDRESS_0         0x40066000
#define NXP_KINETIS_I2C_40066000_SIZE_0         0x1000

erwango, 2017-11-20:
@galak , @agross-linaro: 
We are adding "#size-cells = <0>;", for I2C on both NXP and ST for I2C
This is not done for UART and UART generates the correct define (_SIZE_0) (inheriting from armv6/7.dtsi #size-cells = <1>;)

Should we have same size-cells for both UART and I2C? 
If yes, then one of these above is wrong. And then either dts either script should be updated.
(If not I don't really see why).

Anyway, if you can confirm the exact configuration, I'll have a deeper look to the 'reg' generation algo.

nashif, 2018-02-20:
is this still an issue?

galak, 2018-02-20:
Yes, looking at it now

PR #6282 (2018-02-21): scripts: extract_dts_inculdes: Fix reg prop size/addr cell calculation
When we determine the address and size cells to determine how to parse a
reg property, we need to stop at the parent of the node, not at the node
itself.  If we have #address-cells or #size-cells in a node its meant
for the children of that node and not the node itself.

Fixes #4568
commit 451fc368 (2018-02-21): scripts: extract_dts_inculdes: Fix reg prop size/addr cell calculation

When we determine the address and size cells to determine how to parse a
reg property, we need to stop at the parent of the node, not at the node
itself.  If we have #address-cells or #size-cells in a node its meant
for the children of that node and not the node itself.

Fixes #4568
closed 2018-02-21 by galak