zephyrproject-rtos/zephyr #4569: LoRa:  support LoRa

guo-mkcooc, 2017-10-26:
Hi, 
Recently I do something about LoRa in STM32, i would like to transplant  SX127x radio drivers for zephyr to support LoRa.
Does zephyr have plan to support LoRa ?   thanks.

tbursztyka, 2017-10-26:
I haven't heard about someone working on it.

If you are proposing yourself to bring such support to Zephyr, it will be nice.
I don't know about SX127x, it is a Full or Soft MAC chip?

guo-mkcooc, 2017-10-27:
well,  Thanks for your reply,  I'll try it.

it's a wireless transceivers,  providing spi interface access to the configuration register.

closed 2017-10-27 by guo-mkcooc

rodrigopex, 2018-01-17:
Guys, my company need this. We are about to start to implement the sx127x lib for Zephyr. 
Any tips beyond the docs in how contributing? For this, we need SPI, RTC and GPIO basically.

tbursztyka, 2018-01-17:
sx127 is lora radio chip as far as I can tell. So it lacks all the above: the loraWAn MAC layer. 

You will need this to get a generic interface for the users and also a generic device driver API for the lora chip as well so not only sx127 chip but others would be able to work transparently.
But I guess that goes to #5436

rodrigopex, 2018-01-17:
@tbursztyka  I think I could not say it in a right way. Yes, sx127x is a chip and we will work to make this available to be used in Zephyr. The LoRaWAN porting is a big portion of this work. I would like to use generic drivers to make it available to other board and architectures.

tbursztyka, 2018-01-17:
Thus why I bring this LoRaWAN into the discussion: without a MAC layer, the driver will be pretty useless.

About making a generic driver: Zephry provides generic APIs for spi, gpio and rtc. Your driver will be using those so it will be de-facto generic.

miyatsu, 2018-01-17:
Hi, @tbursztyka 
Recently, I'm working on the LoRaWAN too. Check my local branch here: https://github.com/miyatsu/zephyr/commits/lorawan

I want porting the LoRaWAN-node into zephyr same as you do @rodrigopex 

My design is make the lorawan just like bluetooth in zephyr, become a independent component. The patch is rough, please don't mind.

rodrigopex, 2018-01-17:
@miyatsu as you have already started to work I could help you. What do you think?
The patch seems to be in a good direction. We could create a channel at IRC.

tbursztyka, 2018-01-17:
@miyatsu Since you are porting ready-made code, you will have to put it into ext/lib  (or ext/subsys maybe). Then subsys/lorawan would be the place where it will expose zephyr-ready API (i.e.: no camel-case names, linux code style base code, zephyr types, etc...)

miyatsu, 2018-01-18:
@tbursztyka Actually, I want the zephyr to support LoRaWAN and running OK just for me, I didn't have a plan to porting LoRaMAC to zephyr and merge to the master branch.

miyatsu, 2018-01-18:
@rodrigopex ```We could create a channel at IRC.``` Sorry, I don't know any thing about IRC.

I will update my work at here https://github.com/miyatsu/zephyr/commits/lorawan you can open a pull request and we can discuss here.

tbursztyka, 2018-01-18:
@miyatsu 
> @tbursztyka Actually, I want the zephyr to support LoRaWAN and running OK just for me, I didn't have a plan to porting LoRaMAC to zephyr and merge to the master branch.

Your work seem to be nice to be merged though, as long as you put it into ext/lib. If you don't want to work on the zephyr L2 part, your port of the lorawan into ext/lib would be quite nice already. Then somebody else could use it to do the rest.

mtuxpe, 2019-04-03:
@tbursztyka  What is the suitable directory for LoRa node?
https://github.com/Lora-net/LoRaMac-node
/ext/hal/semtech
/ext/lib/semtech

Ahsan-Averos, 2021-06-01:
if someone integrate nrf52 with Sx1262, then please share code here.
Thanks