zephyrproject-rtos/zephyr #4578: tests/net/socket/udp/testcase.yaml#test : Build failed on esp32

rrahulkx, 2017-10-27:
when I am running this testcase.I am able to re -produce this failure and log pasted below --

**Build procedure :--**

tests/net/socket/udp/testcase.yaml#test @ esp32:xtensa COMPLETION failed

Toolchain: ZEPHYR_GCC_VARIANT=espressif

Commit ID                                                            Branch          URL
---------------------------------------- --------------- -------------------
dc3f2455d7a69db2e1b53fd24f4594d413aff867 master          https://github.com/zephyrproject-rtos/zephyr.git

Detailed description
====================
Automated test run failed.

Output (the target might have produced none; GDB will be needed at this point):

configure passed

build passed: 'mkdir -p outdir-bxud-6k2x-esp32; outdir-bxud-6k2x-esp32/.config'

build passed: 'make -j -C zephyr/tests/net/socket/udp/ 
EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:6k2x 
BOARD=esp32 ARCH=xtensa  
O=outdir-bxud-6k2x-esp32 initconfig'

build output: Using zephyr/boards/xtensa/esp32/esp32_defconfig as base
build output: Merging prj.conf
build output: Merging outdir-bxud-6k2x-esp32/boot_config.conf
build output: # configuration written to .config

build failed: 'make -j -C zephyr/tests/net/socket/udp 
EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:6k2x BOARD=esp32 ARCH=xtensa  
O=outdir-bxud-6k2x-esp32'

build output: /opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-strtol.o):(.literal+0x8): undefined reference to `__getreent'
build output: /opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-strtol.o): In function `strtol':
build output: /home/ivan/e/crosstool-NG/.build/src/newlib-2.2.0/newlib/libc/stdlib/strtol.c:223: undefined reference to `__getreent'
build output: collect2: error: ld returned 1 exit status
build output:zephyr/Makefile:876: recipe for target 'zephyr_prebuilt.elf' failed
build output: make[3]: *** [zephyr_prebuilt.elf] Error 1
build output: Makefile:178: recipe for target 'sub-make' failed
build output: make[2]: *** [sub-make] Error 2
build output:zephyr/Makefile.inc:82: recipe for target 'all' failed

build failed trace: 'make -j -C zephyr/tests/net/socket/udp 
EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:6k2x 
BOARD=esp32 
ARCH=xtensa  O=outdir-bxud-6k2x-esp32\'

build failed
COMPLETION failed

rrahulkx, 2017-11-24:
As per  Zephyr Commit ID:07e1a64e77e15180721c56993a19bce473ff4365

This TC passed on **frdm_k64f** board.

But it failed on below boards at run time.

1)**arduino101**

console output

Running test suite socket_udp
===================================================================
starting test - test_send_recv_2_sock

2)**sam_e70_xplained**

console output:

Running test suite socket_udp
===================================================================
starting test 

- test_send_recv_2_sock
***** MPU FAULT *****
  Executing thread ID (thread): 0x20400730
  Faulting instruction address:  0x406100f4
  Instruction Access Violation
Fatal fault in thread 0x20400730! Aborting.qemu_x86**

3) **quark_se_c1000**

console output:

Running test suite socket_udp
===================================================================
starting test - test_send_recv_2_sock

nashif, 2018-02-09:
same issue as #4705

closed 2018-02-09 by nashif