zephyrproject-rtos/zephyr #4592: Coverity issue seen with CID: 178236

Shilpashree3lc, 2017-10-27:
Static code scan issues seen in File: /drivers/ieee802154/ieee802154_mcr20a.c
 Category: Memory - corruptions
 Function: _mcr20a_read_burst
 Component: Drivers
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

jfischer-phytec-iot, 2017-10-27:
already taken into account in the commit 114d671e24e47fe0c56f319d368ff07642d2e13c

dleach02, 2017-10-27:
@jfischer-phytec-iot , I left a comment on your PR that it doesn't actually fix this bug. The length test is an "off-by-one" problem and needs to have ">=" instead of ">". I have a PR to fix this as well: #4575 which did a similar change but also fixed the condition test.

jfischer-phytec-iot, 2017-10-27:
@dleach02 yes, I have overlooked it, since you have already opened a PR, please pick 
114d671 to your PR and add a commit that fixes this issue, thx

jfischer-phytec-iot, 2017-10-30:
I think Coverity makes a mistake here, I will leave it open until someone confirms.

dleach02, 2017-10-30:
sounds good. I've been told that Coverity is run manually about once a week so lets see what the next report posts.

nashif, 2017-11-15:
this one already fixed

closed 2017-11-15 by nashif