zephyrproject-rtos/zephyr #4597: tests/drivers/ipm/testcase.yaml#test :unable to print the Expected output

rrahulkx, 2017-10-27:
when I run this testcase, its unable to print the entire expected output and could not able to run testcase successfully.

The failure report is pasted below :--

tests/drivers/ipm/testcase.yaml#test @ esp32:xtensa COMPLETION failed

Toolchain: ZEPHYR_GCC_VARIANT=espressif

Commit ID                                                           Branch          URL
---------------------------------------- --------------- -------------------
dc3f2455d7a69db2e1b53fd24f4594d413aff867 master          https://github.com/zephyrproject-rtos/zephyr.git

Detailed description
====================
configure passed

build passed: 'mkdir -p outdir-zqod-ckus-esp32; rm -f outdir-zqod-ckus-esp32/.config'

build passed: 'make -j -C tests/drivers/ipm/ EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" KCPPFLAGS=-DTC_RUNID=ci-171026-0712-438:ckus BOARD=esp32 ARCH=xtensa  O=outdir-zqod-ckus-esp32 initconfig'
build output: Using xtensa/esp32/esp32_defconfig as base
build output: Merging tests/include/test.config
build output: Merging prj.conf
build output: Merging outdir-zqod-ckus-esp32/boot_config.conf
build output: # configuration written to .config

build passed: 'make -j -C tests/drivers/ipm 
EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=ci-171026-0712-438:ckus 
BOARD=esp32 ARCH=xtensa  
O=outdir-zqod-ckus-esp32'

build passed

deploy passed

Reset
**Console output**

console output: starting test - Test IPM
console output: ipm_recv0: 'everything is awesome'
console output: ipm_recv0: 'Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor '

nashif, 2017-10-27:
> The failure report is pasted below :--

I do not see any failures

nashif, 2018-02-09:
can't reproduce

closed 2018-02-09 by nashif