zephyrproject-rtos/zephyr #4603: sanitycheck either conceals information from user or spams it

pfalcon, 2017-10-27:
When I run sanitycheck normally, e.g.:

~~~
./scripts/sanitycheck -s tests/net/socket/udp/test
~~~

, majority of time I'm being shown useless information for dozens of seconds if not minutes, like:

~~~
Generating LALR tables
Selecting default platforms per test case
Building testcase defconfigs...
4 tests selected, 30376 tests discarded due to filters
total complete:    0/   4   0%  failed:    0
~~~

It's not ok to tell me that 4 tests selected and 1 of them ran - I want to know *which test is being run now and which finished running so far*. I don't want to spend minutes just to find out that it ran something else than I wanted.

Ok, solution should be simple - just add `-v`. And now I'm greeted with wall of output right on the startup. Apparently, the thing dumps all those 30K of tests which it decided to skip.

All that's not good. I wanted to write this report months ago, but worked it around in a smart way - by avoiding to run sanitycheck.

Maybe it's time to fix it, instead of making people being afraid to use it.

pfalcon, 2017-10-27:
As https://github.com/zephyrproject-rtos/zephyr/pull/4604 explains, the only way to use sanitycheck sanely is with `sanitycheck | cat` (with that patch applied).

So, the best proposal:

1. Make whatever sanitycheck | cat outputs to be the default output.
2. Then add --terse switch or something for current riddly "0/4" output.

Compromise proposal:

1. Leave current output as is (sucks)
2. Make -v output what sanitycheck | cat does.
3. Make spam to be output with higher levels of verbosity.

nashif, 2017-10-27:
yeah, this has been annoying me as well, listing all discarded tests is not really useful, this should be  another flag which enables listing all discarded tests or change of -v behaviour, but a tip for you, you can run 

`./scripts/sanitycheck -T tests/net/socket/udp`

instead, it will only deal with that specific test and ignore everything else...

pfalcon, 2017-10-27:
@nashif : Thanks for the lifehack, good to know, though doesn't help with the output delay issue. Can we merge #4604 in the meantime?

nashif, 2017-10-27:
I am still not sure what is #4604 is good for. The "output delay" you mention....

> majority of time I'm being shown useless information for dozens of seconds if not minutes

This is because you are executing one test but parse and preprocess all tests before that, the -T should really just preprocess the selected test and there should not be any delay unless there are kconfig filters.

nashif, 2017-10-27:
actually just noticed that -s behave the same way as -T, it just runs the selected test in testcase.yaml, so your issue is with -v and showing discarded tests?

pfalcon, 2017-10-27:
> I am still not sure what is #4604 is good for. The "output delay" you mention....

With #4604, I can run "sanitycheck ... | cat" and get the output which I want (which is the names of tests being run, in realtime).

Without #4604, if I run "sanitycheck ... | cat", I get silence, with big stdout-drop when sanitycheck exits.

pfalcon, 2017-10-27:
> so your issue is with -v and showing discarded tests?

Yes, my issue is what this ticket tells - there's either too little or too much output. #4604 is the simplest and quickest partial solution, before further patches are dropped (which will need more discussion unlike #4604 one-liner).

nashif, 2017-10-27:
we do need to run sanitycheck without having to pipe this through cat, your patch just enable a hack instead of fixing the real problem, I did not see a negative effect of your patch yet, but this requires some additional testing and we need to be able to run with -v and get the desired results without the cat

pfalcon, 2017-10-27:
> I did not see a negative effect of your patch yet, but this requires some additional testing

You must be kidding me - what negative effects can be from flushing the stdout stream? It's just that if Python (and not only Python, many libc's do that) detects that stdout file descriptor is `!isatty(fd)`, it turns on bulk buffering instead of the default line buffering, which causes output delay if fd is a pipe, fifo, etc. It would affect CI systems too (being able to watch output on reatime), but e.g. Shippable apparently redirects output using pseudotty which avoids this issue (returns `isatty(fd) == true`). So, only positive effects can be from that change, even though that it's indeed a partial fix, the full fix needs further discussion, e.g. https://github.com/zephyrproject-rtos/zephyr/issues/4603#issuecomment-339951957 proposes 2 choices, and so far nobody acked one over another.

andrewboie, 2017-10-27:
you know, in the time you took to write all this you could have just adjusted sanitycheck's verbose output, as I have just done in #4612

pfalcon, 2017-10-27:
Well, if everyone will just go and throw patches without discussing an issue ... we'll have the situation like we have now (or worse, it being flip-flopping).

The patch looks nice, but my solution was to switch "verbose()" and "debug()" level (then adjust few cases of verbose to debug):

~~~
 def debug(what):
-    if VERBOSE >= 1:
+    if VERBOSE >= 2:
         info(what)
 
 def verbose(what):
-    if VERBOSE >= 2:
+    if VERBOSE >= 1:
         info(what)
~~~

(And yeah, it's not ready to be posted yet, needs more self-review and testing.)

nashif, 2017-10-27:
> You must be kidding me

no I am not, as I said, I am fine with the change, but I did not spend time looking at where the flush is being done and why. There is a lot of magic that happens in sanitycheck and we better not break it just because....

pfalcon, 2017-10-28:
@nashif : Sure, take your time, that's what reviews are for. But in all fairness, I can't imagine a more obvious and safe change than that. If you look at the patch, 2 lines below the flush is being done to the log file: https://github.com/zephyrproject-rtos/zephyr/pull/4604/files#diff-2287566453359d41de0ff5c53acf48d6L250 . (Then again, it makes more sense to flush stdout (because it has higher chance of interactivity) than a file).

nashif, 2017-12-25:
resolved

closed 2017-12-25 by nashif