zephyrproject-rtos/zephyr #4606: usb mass storage : config waits for Vendor ID and Product ID from user during building

akhileshupadhyay, 2017-10-27:
While building application for usb mass storage, config is expecting user to pass Vendor ID and Product id. Build keeps waiting until provided with IDs. This creates blockage for automation. 

=================================
Below logs appear during building:

````
Merging prj.conf
#
# configuration written to .config
#
make[1]: Entering directory 'ZEPHYR_BASE'
make[2]: Entering directory 'ZEPHYR_BASE/samples/subsys/usb/mass/outdir/quark_se_c1000_devboard'
  GEN     ./Makefile
scripts/kconfig/conf --silentoldconfig Kconfig
.config:434:warning: symbol value '' invalid for USB_DEVICE_VID
.config:435:warning: symbol value '' invalid for USB_DEVICE_PID
*
* Restart config...
*
*
* USB device stack
*
USB device stack (USB_DEVICE_STACK) [Y/n/?] y
  Sets log level for the USB device stack (SYS_LOG_USB_LEVEL) [1] 1
  USB Vendor ID (USB_DEVICE_VID) [] (NEW)

````

Steps to reproduce:

1. cd   $ZEPHYR_BASE
2. source  zephyr-env.sh
3. cd   $ZEPHYR_BASE/samples/subsys/usb/mass
4. make  BOARD=quark_se_c1000_devboard

Work around:
````
Pass USB_DEVICE_VID and USB_DEVICE_PID to boards/x86/quark_se_c1000_devboard/Kconfig.defconfig
````

finikorg, 2017-10-27:
We do not have VENDOR_ID and PRODUCT_ID assigned for this board, however if you know uark_se_c1000_devboard default values we can put them to this defconfig. BTW, do not you use sanitycheck for automatic build?

nashif, 2017-10-27:
@finikorg this is breaking sanitycheck and we do need to have some default values added in this case, same as with tinyTILE.

nashif, 2017-10-27:
ok, I mixed this up with another issue we had with IDs, but in this case for the arduino 101, they all share the same ID like the tinyTILE, so this needs to be set somewhere, in this case probably in the test framework testing the usb

finikorg, 2017-11-01:
I consider this not a bug but feature. Until we get proper Vendor ID we cannot invent those values and put to the project configuration files. Automation issues are resolved in sample.yaml configurations.

jfischer-phytec-iot, 2017-11-02:
currently it is desired behavior

kumarvikash1, 2017-11-29:
As discussed above, Now VID & PID values are being provided from test framework only. Specific VID & PID will be profided by user, if any. 

We can close this issue as FIXED.

carlescufi, 2018-01-23:
Can we close this as @kumarvikash1 mentioned, or do we need to add any required changes now that there is a "default" Zephyr PID/VID?

finikorg, 2018-02-01:
Closed as fixed

closed 2018-02-01 by finikorg